micropython/micropython #8485: mpremote on Mac OS X: allow env var or option in config file to select default serial port to use

javiercanadillas, 2022-04-03:
I'm using Mac OS X, and I'm struggling to get `mpremote` to do anything more than giving me a REPL.

This is the list of devices I get when trying to connect to an ESP32 DevKitC1 board:

```bash
mpremote connect list
```

```text
/dev/cu.BLTH None 0000:0000 None None
/dev/cu.Bluetooth-Incoming-Port None 0000:0000 None None
/dev/cu.usbserial-0001 0001 10c4:ea60 Silicon Labs CP2102 USB to UART Bridge Controller
```

Issuing `mpremote connect /dev/cu.usbserial-0001` connects to the board and gives me a REPL back. But any other command (`fs`, `run`, `exec`...) won't work as `mpremote` always assume that the device is has to connect to is the first in the previous list (`/dev/cu.BLTH` in my case).

As `mpremote` CLI syntax does not seem to allow specifying both the serial port to use and the action to perform (imagine something like `mpremote -D /dev/cu.usbserial-001 fs ls`), would it be possible to consider a shell env var *à la `ampy`* (or `minicom`) to tell it what serial port to use?

andypiper, 2022-04-03:
I think making it easier for `mpremote` to find a connected board on macOS is a good suggestion. 

You *can* however, get somewhere towards this via the config file. In `~/.config/mpremote/config.py` I have

```python
commands = {
    "wspico": "connect /dev/cu.usbmodem11101",
...
}
```

This lets me do things via `mpremote wspico <action>` which is a bit shorter / less typing.

I'd love to see more work and documentation around `mpremote`, I believe @mattytrentini may have some workflow steps he could share - I'd be happy to write a tutorial or something, but, like you @javiercanadillas I'm still playing at the basics stage of things.

javiercanadillas, 2022-04-03:
Hi @andypiper,

Thank you for your answer and the hints! 

Reading the docs I didn't grasp that I could do something like `mpremote connect /dev/cu.usbserial-0001 fs ls`, but it seems now I should be able to. I guess my issue with it is that for whatever reason, a sequence of `connect` and then another consecutive command like `fs ls` didn't seem idiomatic to me.

By looking at the code I've seen that the env var `XDG_CONFIG_HOME` holds the home for the `.config` directory. I believe this is not documented, nor evident to Mac OS X users. As I'm installing `mpremote` with `pip` under a virtual environment and as I already have all my dotfiles under `~/.config`,  setting the home there seems like the way to go for my workflow.

So:

```bash
mkdir -p $HOME/.config/mpremote
export XDG_CONFIG_HOME="$HOME/.config"
cat <<- EOF > $HOME/.config/mpremote/config.py
commands = {
    "esp32": "connect /dev/cu.usbserial-0001"
}
EOF
```
and then:
```bash
mpremote esp32
```
gives me a REPL. Yay! :-)

But:
```bash
mpremote esp32 fs ls
``` 
fails in `line 340 in enter_raw_repl raise PyboardError("could not enter raw repl")`.

At first, I thought this was because I had a `main.py` running trying to get WiFi connectivity. So I deleted it and tried again, but the `ls` command still fails because the raw repl mode couldn't be entered.

mdaeron, 2022-08-28:
> I think making it easier for `mpremote` to find a connected board on macOS is a good suggestion.
> 
> You _can_ however, get somewhere towards this via the config file. In `~/.config/mpremote/config.py` I have
> 
> ```python
> commands = {
>     "wspico": "connect /dev/cu.usbmodem11101",
> ...
> }
> ```

Thanks for this workaround.

Still, is there any way to tell `mpremote` to ignore certain ports (e.g., `/dev/cu.Bluetooth-Incoming-Port`), or to only consider ports starting with a certain string (e.g., `/dev/cu.usb`)? The ability to autoconnect to the first free device is quite neat when working on systems combining several boards.

thyrrestrup, 2022-08-28:
You could do: /dev/cu.usbmodem* - I use that in an alias.

mdaeron, 2022-08-28:
> You could do: `mpremote connect /dev/cu.usbmodem*` - I use that as an alias.

This seems to break the ability to automatically find the first free device.

If I connect two boards and try `mpremote connect /dev/cu.usb*` twice in a row, the first command works as expected and the second one raises `failed to access /dev/cu.usbmodem1234561 (it may be in use by another program)`.

mdaeron, 2022-08-30:
#9140 aims to fix this issue.

TJC, 2023-03-31:
```
$ mpremote version
mpremote 0.4.0
```

robert-hh, 2023-03-31:
You can chain commands. Does e.g. `mpremote  /dev/cu.usbmodem1234561 ls` work for you?

TJC, 2023-04-01:
Ah! Yes, chaining the commands together does work. I had no idea the connection string carries over from one to the next.

So now I've made a helper script:
```bash
#!/bin/bash
device=$(ls -1 /dev/cu.usb*|head -n 1)
mpremote connect "$device" $*
```

Thanks -- although if it's possible to improve the documention and/or provide a `--device` parameter or `MPREMOTE_DEVICE` env var, that would certainly be appreciated.

TJC, 2023-04-01:
Thanks so much for your quick response with a workaround.