micropython/micropython #8489: S2/S3 RTC_VALID_EXT_PINS definitions

stephanelsmith, 2022-04-04:
Coming around back to this one https://github.com/micropython/micropython/issues/7607

On S2/S3 variants, GPIO0 through GPIO21 are valid RTC pins.  In this example, asserting pin 5 should wake.

```

import esp32
import machine
from machine import Pin
o = Pin(5, Pin.IN, pull=Pin.PULL_DOWN)
esp32.wake_on_ext1(pins = (o,), level = esp32.WAKEUP_ANY_HIGH) #multi-pin

machine.deepsleep(10000)

```

stephanelsmith, 2022-04-04:
Pull request makes S2/S3 port specific definitions.    I didn't include C3 as I'm not familiar with the port.  A casual glance at the datasheet doesn't show any pins with RTC function.
- https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf

S2/S3 data sheets clearly show pin functions with RTC supply (external wake).
- https://www.espressif.com/sites/default/files/documentation/esp32-s2_datasheet_en.pdf
- https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf

Proposed pull request is:
```
#if CONFIG_IDF_TARGET_ESP32S2 || CONFIG_IDF_TARGET_ESP32S3

    #define RTC_VALID_EXT_PINS \
        ( \
        (1ll << 0) | \
        (1ll << 1) | \
        (1ll << 2) | \
        (1ll << 3) | \
        (1ll << 4) | \
        (1ll << 5) | \
        (1ll << 6) | \
        (1ll << 7) | \
        (1ll << 8) | \
        (1ll << 9) | \
        (1ll << 10) | \
        (1ll << 11) | \
        (1ll << 12) | \
        (1ll << 13) | \
        (1ll << 14) | \
        (1ll << 15) | \
        (1ll << 16) | \
        (1ll << 17) | \
        (1ll << 18) | \
        (1ll << 19) | \
        (1ll << 20) | \
        (1ll << 21)   \
        )
    #define RTC_LAST_EXT_PIN 21

#else

    #define RTC_VALID_EXT_PINS \
        ( \
        (1ll << 0) | \
        (1ll << 2) | \
        (1ll << 4) | \
        (1ll << 12) | \
        (1ll << 13) | \
        (1ll << 14) | \
        (1ll << 15) | \
        (1ll << 25) | \
        (1ll << 26) | \
        (1ll << 27) | \
        (1ll << 32) | \
        (1ll << 33) | \
        (1ll << 34) | \
        (1ll << 35) | \
        (1ll << 36) | \
        (1ll << 37) | \
        (1ll << 38) | \
        (1ll << 39)   \
        )
    #define RTC_LAST_EXT_PIN 39

#endif

```

dpgeorge, 2022-04-14:
Done in 988827b85a9b09f1164e101ea2520562dd31634e

closed 2022-04-14 by dpgeorge