micropython/micropython #8491: S3R8 (octal spram configuration)

stephanelsmith, 2022-04-05:
The S3R8 and S3R8V ports have an 8Mb octal SPI PSRAM configuration (the rest of the S3 ports are quad).  See page 20 of data sheet:
https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf

In machine_pin.c, recommending 

```
 #if CONFIG_SPIRAM_MODE_OCT
    {{NULL}, -1}, // 33 FLASH/PSRAM
    {{NULL}, -1}, // 34 FLASH/PSRAM
    {{NULL}, -1}, // 35 FLASH/PSRAM
    {{NULL}, -1}, // 36 FLASH/PSRAM
    {{NULL}, -1}, // 37 FLASH/PSRAM
    #else
    {{&machine_pin_irq_type}, GPIO_NUM_33},
    {{&machine_pin_irq_type}, GPIO_NUM_34},
    {{&machine_pin_irq_type}, GPIO_NUM_35},
    {{&machine_pin_irq_type}, GPIO_NUM_36},
    {{&machine_pin_irq_type}, GPIO_NUM_37},
    #endif

```

This update added into the pull request for s3 specific pin definitions.
https://github.com/micropython/micropython/pull/8490

stephanelsmith, 2022-04-05:
Resolves https://github.com/micropython/micropython/pull/8490

dpgeorge, 2022-04-14:
Fixed by 700535409908687633237e895f3629cb1a419c64

closed 2022-04-14 by dpgeorge