micropython/micropython #8494: rp2: Random lockups on soft reboot if core 1 was running

czietz, 2022-04-05:
Running the following minimal example on the Raspberry Pi Pico:
```python
import _thread
def burn():
    while True:
        pass

_thread.start_new_thread(burn, [])
```
... starts an endless loop on the second CPU core (aka core 1). If a soft reboot via Ctrl+D is then performed, Micropython sometimes locks up to the point where only a hardware reset of the Pico helps. (You might have to try this several times, it does not happen _every_ soft reboot.)

I've traced this to an endless wait for the spinlock in [_osal_q_lock](https://github.com/hathach/tinyusb/blob/4bfab30c02279a0530e1a56f4a7c539f2d35a293/src/osal/osal_pico.h#L124). I think, this is because the TinyUSB task is called on both cores via `MICROPY_HW_USBDEV_TASK_HOOK`:
https://github.com/micropython/micropython/blob/587339022689187a1acbccc1d0e2425a67385ff7/ports/rp2/mpconfigport.h#L232

Therefore, I think this is what happens: During a soft reboot, the core 1 is reset:
https://github.com/micropython/micropython/blob/587339022689187a1acbccc1d0e2425a67385ff7/ports/rp2/mpthreadport.c#L50
If this happens at an unfavorable moment, when core 1 holds the Tiny USB spinlock, said lock is never released and core 0 hangs indefinitely when trying to acquire the lock the next time.

At first glance, only calling the TinyUSB task on core 0:
```c
#define MICROPY_HW_USBDEV_TASK_HOOK extern void tud_task(void); if (get_core_num() == 0) tud_task();
```
... seems to fix the problem, as I cannot observe any random lockups.

However, I'm not familiar enough with Micropython's internals to know if there are situations in which the task needs to run on core 1, too. Therefore, I did not propose this tentative fix as a PR.

czietz, 2022-04-06:
Note that there might be yet _another_ deadlock situation: When core 1 is reset while it is running its local garbage collection, the GC mutex might be held indefinitely. That would become an issue as soon as core 0 runs GC during soft reboot:
https://github.com/micropython/micropython/blob/587339022689187a1acbccc1d0e2425a67385ff7/ports/rp2/main.c#L162

Perhaps some synchronization needs to be added before stopping/resetting core 1 to ensure that all shared locks are free.

jimmo, 2022-06-29:
Thank you @czietz for the very detailed report and investigation!! I was able to repro both issues. See #8835 and #8836

dpgeorge, 2022-07-12:
Fixed by 797a83ac3ef8f9bbf67faf59da665b5509db428b and daff597753e699bb9f485e69dea98f58b0c1432d

closed 2022-07-12 by dpgeorge