micropython/micropython #8498: GC: automatic collect on import

Andrei-Pozolotin, 2022-04-06:
1) a well known workaround to reduce memory fragmentation on start is to intersperse module `import` with `gc.collect()`

2) this is a feature request to provide a mechanism (a global flag?) to activate "automatic gc.collect on import", meaning that 
each import:
```python
import XYZ
```
effectively becomes:
```python
import XYZ
gc.collect()
```

stinos, 2022-04-06:
Should probably be 'gc.collect() if not previously imported'?

Andrei-Pozolotin, 2022-04-06:
yes

peterhinch, 2022-04-06:
Note that `gc.collect()` on platforms with SPIRAM can take up to 300ms even when there is no garbage to collect.

Andrei-Pozolotin, 2022-04-06:
good to know

jonnor, 2024-09-08:
I think that explicit is better than implicit here. Using gc.collect() manually can be used in those cases where it is needed.
Unless core maintainers disagree, I propose we close this.

dpgeorge, 2024-09-09:
This would actually be a good feature to have.

Actually, it can already be mostly achieved by setting a very low GC threshold before the bulk of the imports occur, eg: `gc.threshold(1000)`.