micropython/micropython #8506: Corruption of the file system with file or directory deletion during a loop with ilistdir

remibert, 2022-04-08:
It took me a very long time to find the cause of sporadic corruption. 
Corruption can occured on an sdcard and on the internal file system (littlefs or fat).
If you destroys a directory or a file in the loop of an ilistdir, some time after (not immediately) you can get a crashes, or filesystem corruption (in the case of the sdcard it is frequently necessary to reformat). 
If you store the deletion in a list, and do the destruction after the loop, the problem no longer occurs.
Could you add documentation on the ilistdir function this limitation.

robert-hh, 2022-04-08:
can you show a sample script and test set-up for this issue? Otherwise it's hard to replicate. If it is a bug, it must be fixed instead of just telling, that something may go wrong.

remibert, 2022-04-08:
Not in a simple way, I had the problem very sporadically, until I decided to save images in the flash memory of the esp32. As it quickly reached saturation of the available space, it cleaned up the old images more frequently, and just after it crashed (less than 5 minutes to crash in lfs2.c or freeze the esp32).

I corrected it here line 234 :
https://github.com/remibert/pycameresp/blob/main/modules/lib/motion/historic.py <https://github.com/remibert/pycameresp/blob/main/modules/lib/motion/historic.py>

See recent changes on remove_files method :
https://github.com/remibert/pycameresp/commit/98200199bdef8432eef144d4cc76fa417a10ae64

The difficulty it is embedded in a large number of async functions (many servers), not easily extractable.

I produce it with an esp32cam, esp32one with camera, M5 camera, with micropython firmware modified to support cameras.

The changes made to the micropython firmware have no connection with this crash, I have cameras that have been working for months, without any particular problems of file system corrupted.

robert-hh, 2022-04-08:
At least I understand from you description, that it is an ESP32 with lfs2, with no filesystem exposed to a PC. And saving files frequently until the free space is exhausted leads to that error. What is the size of the files you store to the esp32?

remibert, 2022-04-08:
Here an example of directory content (with size) :
2000/01/01  00:00:00         [data/2022]
2000/01/01  00:00:00         [data/2022/03]
2000/01/01  00:00:00         [data/2022/03/29]
2000/01/01  00:00:00         [data/2022/03/29/16h50]
2022/03/29  16:53:34    383B data/2022/03/29/16h50/2022-03-29_16-52-22 Id=33 D=20.json
2000/01/01  00:00:00         [data/2022/04]
2000/01/01  00:00:00         [data/2022/04/06]
2000/01/01  00:00:00         [data/2022/04/06/19h15]
2022/04/06  19:18:48    385B data/2022/04/06/19h15/2022-04-06_19-18-08 Id=105 D=101.json
2022/04/06  19:18:53  54.02K data/2022/04/06/19h15/2022-04-06_19-18-12 Id=110 D=105.jpg
2000/01/01  00:00:00         [data/2022/04/07]
2000/01/01  00:00:00         [data/2022/04/07/14h50]
2022/04/07  14:51:46  56.11K data/2022/04/07/14h50/2022-04-07_14-51-38 Id=2279 D=16.jpg
2022/04/07  14:51:49    385B data/2022/04/07/14h50/2022-04-07_14-51-38 Id=2279 D=16.json
2000/01/01  00:00:00         [data/2022/04/07/16h10]
2022/04/07  16:11:48  50.70K data/2022/04/07/16h10/2022-04-07_16-11-36 Id=59 D=6.jpg
2022/04/07  16:11:50    382B data/2022/04/07/16h10/2022-04-07_16-11-36 Id=59 D=6.json
2022/04/07  16:11:54  49.99K data/2022/04/07/16h10/2022-04-07_16-11-37 Id=60 D=81.jpg
2022/04/07  16:11:57    383B data/2022/04/07/16h10/2022-04-07_16-11-37 Id=60 D=81.json
2022/04/07  16:12:03  53.27K data/2022/04/07/16h10/2022-04-07_16-11-39 Id=62 D=134.jpg
2022/04/07  16:12:05    384B data/2022/04/07/16h10/2022-04-07_16-11-39 Id=62 D=134.json
2022/04/07  16:12:11  53.20K data/2022/04/07/16h10/2022-04-07_16-11-41 Id=64 D=130.jpg
2022/04/07  16:12:13    384B data/2022/04/07/16h10/2022-04-07_16-11-41 Id=64 D=130.json
2022/04/07  16:12:20  53.42K data/2022/04/07/16h10/2022-04-07_16-11-45 Id=68 D=130.jpg
2022/04/07  16:12:24    384B data/2022/04/07/16h10/2022-04-07_16-11-45 Id=68 D=130.json
2022/04/07  16:12:31  53.36K data/2022/04/07/16h10/2022-04-07_16-12-01 Id=71 D=131.jpg
2000/01/01  00:00:00         [data/2022/04/07/16h15]
2022/04/07  16:18:33  54.13K data/2022/04/07/16h15/2022-04-07_16-17-53 Id=422 D=9.jpg
2022/04/07  16:18:36    383B data/2022/04/07/16h15/2022-04-07_16-17-53 Id=422 D=9.json
2022/04/07  16:18:41  54.57K data/2022/04/07/16h15/2022-04-07_16-18-23 Id=426 D=23.jpg
2022/04/07  16:18:43    384B data/2022/04/07/16h15/2022-04-07_16-18-23 Id=426 D=23.json
2000/01/01  00:00:00         [data/2022/04/07/16h20]
2022/04/07  16:20:53  53.91K data/2022/04/07/16h20/2022-04-07_16-20-43 Id=44 D=10.jpg
2022/04/07  16:20:55    383B data/2022/04/07/16h20/2022-04-07_16-20-43 Id=44 D=10.json
2022/04/07  16:20:59  53.83K data/2022/04/07/16h20/2022-04-07_16-20-44 Id=45 D=9.jpg
2022/04/07  16:21:02    382B data/2022/04/07/16h20/2022-04-07_16-20-44 Id=45 D=9.json
2022/04/07  16:21:06  54.03K data/2022/04/07/16h20/2022-04-07_16-20-47 Id=47 D=12.jpg
2022/04/07  16:21:08    383B data/2022/04/07/16h20/2022-04-07_16-20-47 Id=47 D=12.json
2022/04/07  16:21:30  53.79K data/2022/04/07/16h20/2022-04-07_16-21-15 Id=63 D=10.jpg
2022/04/07  16:21:32    383B data/2022/04/07/16h20/2022-04-07_16-21-15 Id=63 D=10.json
2022/04/07  16:21:36  53.85K data/2022/04/07/16h20/2022-04-07_16-21-17 Id=64 D=16.jpg
2022/04/07  16:21:39    383B data/2022/04/07/16h20/2022-04-07_16-21-17 Id=64 D=16.json
2022/04/07  16:21:43  54.69K data/2022/04/07/16h20/2022-04-07_16-21-18 Id=65 D=17.jpg
2000/01/01  00:00:00         [data/2022/04/07/16h25]
2022/04/08  21:13:41  47.09K data/2022/04/07/16h25/2022-04-07_16-25-06 Id=21 D=145.jpg
2022/04/08  21:13:43    384B data/2022/04/07/16h25/2022-04-07_16-25-06 Id=21 D=145.json

The remove_files delete older files in data/year/month/day/hour, if the hour directory is empty, then it remove all top directories empty recursively. 
Two type of files stored, jpg and json files.

dpgeorge, 2022-04-10:
I think this is a duplicate (same root cause) as #6568.

dpgeorge, 2022-09-13:
Fixed by 4e0964b59f44b25ebaa5239f9ea4273b804ebe64

closed 2022-09-13 by dpgeorge