micropython/micropython #8507: Is the parser technology gap between MP and CPython going to pose a serious problem?

RayaneCTX, 2022-04-08:
I feel that MicroPython's desire to keep up with CPython's language features is coming into serious conflict with the fact that CPython's parser technology is superior (PEG parser since Python 3.9). I believe that that fact alone will increasingly and disproportionately contribute to a widening gap between MicroPython and CPython. For example, I have recently been looking at implementing two syntax features: #8479 (function keyword syntax restriction) and #8503 (decorator syntax relaxation), and really, both of these changes should have almost exclusively been grammar changes, and that is quite far from the case. I know these are a lot of moving parts and variables to consider; I just feel that way, and I am willing to believe that the symptoms I am noticing can be solved through cleaner, more modular code rather than a parser overhaul. My concern is as follows: how effectively can we keep up with CPython's syntax features while mitigating the "spaghettification" of our code?

What do you feel? Too early to tell? Chasing ghosts? Or are we indeed sitting on a timed bomb?

For reference, https://github.com/micropython/micropython/issues/7847#issuecomment-944317777 discusses a bit why a PEG parser wouldn't be practical for an embedded system.

dpgeorge, 2022-04-10:
We have not yet hit any real issues with the current MicroPython parser.  It has been working well since day 1.  If it were to be changed it would be a big undertaking and a lot of things must be considered, for example code size, heap memory use, C stack use and performance.

At this stage I think it's too early to tell if it needs to be changed to keep up with CPython features.  In the end maybe MicroPython doesn't add structural pattern matching or other new syntax features that are too complicated to implement.

jonnor, 2024-09-08:
Seems we are doing OK so far! Specific issues can be opened if/when we hit bottlenecks in terms of parser type.

closed 2024-09-08 by jonnor