micropython/micropython #8509: Core pinning -> pin.irq handler not called

stephanelsmith, 2022-04-08:
Consolidating (and closing) previous issue.

Issue found in esp32 s3.  Issue is observable from commit 357078504d0d2c3a3f06ec4bd69e0ab3bcf9e016 through current master.
**357078504 esp32: Pin MicroPython to core 1 again.**

Using GENERIC_S3 board, pin.irq handler is not be called.   Issue seen across multiple s3 boards, not observable on s2 boards.

Reproducable with.
```
from machine import Pin
p = Pin(5, Pin.IN, pull=Pin.PULL_DOWN)
p.irq(lambda pin:print(pin))
```

Deep sleep wake events **are** working correctly.
```
o = Pin(5, Pin.IN, pull=Pin.PULL_DOWN)
esp32.wake_on_ext1(pins = (o,), level = esp32.WAKEUP_ANY_HIGH) #multi-pin
```

Switching to unicore mode, pin.irq handler is called.
**CONFIG_FREERTOS_UNICORE=y**

stephanelsmith, 2022-05-20:
Bumping this.  This is a bit beyond my domain understanding.  I'd be happy to look deeper, any clues as to why the irq could be firing on the wrong core?

jbecknyc, 2022-05-31:
This is an esp-idf bug, fixed in 4.4.1.   See https://github.com/espressif/esp-idf/issues/7885.  

Quick fix for idf4.4 is to remove the if statement on line 146 of $IDF_PATH/components/hal/esp32s3/include/hal/gpio_ll.h so the assignment on the following line is unconditional.

stephanelsmith, 2022-06-22:
Wow, that's exactly it!!  
1) I performed the edit as indicated and working as expected.
2) Upgrade to v4.4.1 (looks like there are a number of important updates here for s3, https://github.com/espressif/esp-idf/releases/tag/v4.4.1), and confirming also working as expected!

Very very much appreciate it.  You most definitely ROCK my universe!

What's the protcol on something like this.  Should this issue be closed or left open?  I'm sure there may well be other wary travelers who may encouter this as well.

brianwyld, 2023-09-01:
thanks for logging this issue and the fix - helped me fix my project (hacked the 4.4, as the 4.4.1 update caused me more issues but due to git...)

projectgus, 2023-09-05:
>  What's the protcol on something like this. Should this issue be closed or left open? I'm sure there may well be other wary travelers who may encouter this as well.

I think it was good to keep this open while the issue was present in the IDF version used by MicroPython on master branch.

MicroPython updated to IDF v4.4.2 in 034502bc72e8d1b6371ce2970fdba4b7f4f41710 and v5.0.2 in e4650125b88a35f074097f16d84a8f49bd22ac06 so I'm assuming this is safe to close out now.

closed 2023-09-05 by projectgus