micropython/micropython #8510: ESP32 : Support for WiFi LR mode

AmirHmZz, 2022-04-08:
According to [espressif docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#long-range-lr) there is WiFi protocol which is called **LR Mode** ( stands for Long Range ) : 

> Long Range (LR) mode is an Espressif-patented Wi-Fi mode which can achieve a one-kilometer line of sight range. It has better reception sensitivity, stronger anti-interference ability and longer transmission distance than the traditional 802.11b mode.

 LR Mode can be enabled by calling `esp_wifi_set_protocol(ifx, WIFI_PROTOCOL_LR)`. I just wanted to ask does micropython port for esp32 supports this mode?

glenn20, 2022-05-04:
The LR mode is not supported in the micropython releases, but it is supported in PR #6515. The espnow patches add support for:

```
w0 = network.WLAN(network.STA_IF)
w0.config(protocol=network.MODE_LR)
```

Documentation for LR mode is available at:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-protocol-mode

NOTE: I believe you may need to set this config option BEFORE calling w0.active(True) (ie. before esp_wifi_start() is called).

AmirHmZz, 2022-05-04:
@glenn20 thanks for your attention and hard work on #6515. I'm just checking that PR everyday because our team really needs that to be merged. I will close this issue thanks to your complete answer.

closed 2022-05-04 by AmirHmZz