micropython/micropython #8512: Incredible

hierosme-nova, 2022-04-09:
Hello, 

My trouble is may be is a simple a documentation trouble.
I.m certainly not the only one in totally pain just for a TDD workflow, CI/CD with MicroPython + Third PArty Module.

All my trouble come from the fact i need a OS module it know what is environnement variable (my bad i know), just for it small thing it require to me a anormal amount of work, The result is simple: A almost unsolvable developpement with TDD .

May be if someone take a moment to explain to Devllopper's:
How create a project structure.
How include libs inside that internal structure
How deal with file structure
How update they thrid-party modules during the developpement phase
How ceate a tests
How deport test outside , like a CI/CD
How build images devices
How simulate tests on many virtual device (qemu or other)

Be sure have libraries like OS it not respect the signature of the original lib have terrible impact ... on a workflow like that 

OK, that a fact, OS from MicroPython is like that, then next step ?
How ????

May be we just have to write they documentations by our self, but all i found is interactive HelloWorld, and nothing about Workflow, requirement, TDD, CI/CD , Virtual Machine, Compilation for Unix/Port it all ready include module.

Im no cross, i'm just without documenation

Regards

hierosme-nova, 2022-04-09:
In case it help any one with the same Toolchain trouble:

https://www.digikey.com/en/maker/projects/micropython-basics-loading-modules/9560902ac9654f7e91f098e723d097ad
https://learn.adafruit.com/building-and-running-micropython-on-the-esp8266/build-firmware
https://github.com/pfalcon/pycopy-lib

dpgeorge, 2022-04-09:
This is an issue tracker.  For support requests you need to ask at https://forum.micropython.org.  And you need to describe clearly what you need help with and how you have already tried to solve the problem.

closed 2022-04-09 by dpgeorge

hierosme-nova, 2022-04-09:
Hi @dpgeorge ,

I understand, it look like a forum question, but my issue conserne need of documentation.

dpgeorge, 2022-04-09:
Your suggestions are very broad.  It would be more helpful to say something like "I tried to do this thing but I ran into this and that issue and here is how I solved it in the end, and I think something along this lines can be added to the docs to help others because it's a common problem".