micropython/micropython #8521: samd: USB interface locks up when input sybols arrive too fast.

robert-hh, 2022-04-11:
There is a USB interface issue, that the USB interface locks up if input symbols arrive too fast AND are echoed back. In the master branch, this is hard to replicate. The Issue got evident with PR8520, where the issues can re reproduced easily.
When the USB interface seems to be locked up, the board continues to work. Analysis in the Debugger shows, that input symbols still arrive and are processed. Output to commands will be created and the function mp_hal_stdout_tx_strn() is called, which in turn calls tud_cdc_write() and tud_cdc_write_flush(). Only the data does not arrive at the connected PC/USB interface.

robert-hh, 2022-04-19:
While the USB interface seems to be locked, REPL output still goes to UART for a while, but locks up later as well, probably because the USB output buffer is filled up.
When REPL input from UART is used, all works well. No lock-up.

robert-hh, 2022-06-27:
It looks like I found the culprit in the tinyusb lib. Making the change below the error is gone. Now I have to make  a PR and convince @hathach to change it, and then make the matching PR for the MicroPython repository.
```
diff --git a/src/portable/microchip/samd/dcd_samd.c b/src/portable/microchip/samd/dcd_samd.c
index 526a6fb63..68aa5ab75 100644
--- a/src/portable/microchip/samd/dcd_samd.c
+++ b/src/portable/microchip/samd/dcd_samd.c
@@ -271,14 +271,14 @@ bool dcd_edpt_xfer (uint8_t rhport, uint8_t ep_addr, uint8_t * buffer, uint16_t
   {
     bank->PCKSIZE.bit.MULTI_PACKET_SIZE = total_bytes;
     bank->PCKSIZE.bit.BYTE_COUNT = 0;
-    ep->EPSTATUSCLR.reg |= USB_DEVICE_EPSTATUSCLR_BK0RDY;
-    ep->EPINTFLAG.reg |= USB_DEVICE_EPINTFLAG_TRFAIL0;
+    ep->EPSTATUSCLR.reg = USB_DEVICE_EPSTATUSCLR_BK0RDY;
+    ep->EPINTFLAG.reg = USB_DEVICE_EPINTFLAG_TRFAIL0;
   } else
   {
     bank->PCKSIZE.bit.MULTI_PACKET_SIZE = 0;
     bank->PCKSIZE.bit.BYTE_COUNT = total_bytes;
-    ep->EPSTATUSSET.reg |= USB_DEVICE_EPSTATUSSET_BK1RDY;
-    ep->EPINTFLAG.reg |= USB_DEVICE_EPINTFLAG_TRFAIL1;
+    ep->EPSTATUSSET.reg = USB_DEVICE_EPSTATUSSET_BK1RDY;
+    ep->EPINTFLAG.reg = USB_DEVICE_EPINTFLAG_TRFAIL1;
   }
 
   return true;
```

dpgeorge, 2022-06-28:
Great work!  Let's hope it gets fixed in upstream TinyUSB soon.

robert-hh, 2022-06-28:
Thanks. I made a PR, but for the current master  branch. The SAMD port works with that branch. So either we could work with different commits for the various ports, or we switch them all.

dpgeorge, 2022-06-28:
It's too difficult to use different TinyUSB commits for different ports... would be best to wait for the next release of TinyUSB.

robert-hh, 2022-06-28:
There is already a new TinyUSB version. MicroPython uses version 0.12.0. The master branch is beyond the  version 0.13.0 tag. And the RP2 branch does not build with that version. Two errors:

a) an undefined C Macro: `PICO_SHARED_IRQ_HANDLER_HIGHEST_ORDER_PRIORITY`
b) a missing function: `undefined reference to "tud_task"`

So using a new TinyUSB version requires rework of some ports.

dpgeorge, 2022-06-29:
ESP32 uses it's own version of TinyUSB in the IDF, so no need to worry about that.

There are no stm32 boards using TinyUSB.

I have nRF boards that I can test.

robert-hh, 2022-06-29:
Thanks. Ha Thach mentioned that he will publish a release (v1.x?) of TinyUSB in the next days.

robert-hh, 2022-07-19:
Fixed with the tinyusb update.

closed 2022-07-19 by robert-hh