micropython/micropython #8524: esp32-s3 REPL

ruihua-code, 2022-04-12:
Board : ESP32-S3-12K-kit
System : mac OSX
Firmware : https://micropython.org/download/GENERIC_S3/

```bash
picocom -b 115200 /dev/cu.usbserial-1410
```

Ctrl+C not entry REPL environment:

[image]

ampy Do not use，not entry shell input mode

```bash
ampy -b 115200 -p /dev/cu.usbserial-1410 -d 1 ls
ampy -b 115200 -p /dev/cu.usbserial-1410 -d 1 run main.py
ampy -b 115200 -p /dev/cu.usbserial-1410 -d 1 put main.py
```
[image]

Please help. Thank you

dpgeorge, 2022-04-12:
I think this is the same as #8418: the REPL on the board is on the UART and not the native USB.

Do you have a link for this board, to its documentation?

ruihua-code, 2022-04-13:
> I think this is the same as #8418: the REPL on the board is on the UART and not the native USB.
> 
> Do you have a link for this board, to its documentation?

#8418 I have tried this method and the problem remains

doc link: (ESP-S3-12K):https://docs.ai-thinker.com/en/esp32s3

dpgeorge, 2022-04-13:
According to those schematics, the ESP-S3-12K does have a USB-UART convertor.  So it is the same issue as #8418, and the fix is to enable UART as well as USB on those generic S2/S3 firmware builds.

ruihua-code, 2022-04-13:
> According to those schematics, the ESP-S3-12K does have a USB-UART convertor. So it is the same issue as #8418, and the fix is to enable UART as well as USB on those generic S2/S3 firmware builds.

* 新固件构建
```
make erase
make deploy
```
 **I have tried this method and the problem remains**！！！

**What else can I do?**

**Arduino can upload files normally, but I prefer MicroPython.**

[image]

dpgeorge, 2022-04-13:
Please try this patch:
```diff
--- a/ports/esp32/main.c
+++ b/ports/esp32/main.c
@@ -90,6 +90,7 @@ void mp_task(void *pvParameter) {
     #endif
     #if CONFIG_USB_ENABLED
     usb_init();
+    uart_stdout_init();
     #elif CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG
     usb_serial_jtag_init();
     #else
diff --git a/ports/esp32/mphalport.c b/ports/esp32/mphalport.c
index 41e6e6ec0..d0db57d78 100644
--- a/ports/esp32/mphalport.c
+++ b/ports/esp32/mphalport.c
@@ -109,6 +109,7 @@ void mp_hal_stdout_tx_strn(const char *str, size_t len) {
     }
     #if CONFIG_USB_ENABLED
     usb_tx_strn(str, len);
+    uart_stdout_tx_strn(str, len);
     #elif CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG
     usb_serial_jtag_tx_strn(str, len);
     #else
```

ruihua-code, 2022-04-13:
> 

[image]

closed 2022-04-13 by ruihua-code

ruihua-code, 2022-04-13:
> Please try this patch:
>
> ```diff
> --- a/ports/esp32/main.c
> +++ b/ports/esp32/main.c
> @@ -90,6 +90,7 @@ void mp_task(void *pvParameter) {
>      #endif
>      #if CONFIG_USB_ENABLED
>      usb_init();
> + [...]

**Very successful, cool  perfect**

Thank you for help.

[image]

dpgeorge, 2022-04-13:
Great!  Thanks for testing.  This patch (or something similar) will go in master.

dpgeorge, 2022-04-14:
Fixed by fe9eaf12f391232f195ac052005ac2b852409dd6