micropython/micropython #8527: ESP32: Support I2S-driven ADC sampling

oclyke, 2022-04-13:
The [ESP32 I2S peripheral](https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32/api-reference/peripherals/i2s.html) supports an interesting feature: it can be used to push ADC samples into DMA memory. Currently the ESP32 I2S interface does not allow configuration of the peripheral in this way, though it would only require a little effort.

There is not much configuration available. Besides sample rate and number of bits (already included in the I2S module) 
* which ADC peripheral to use
* which ADC channel to use

Proposal:
* create a class similar to the I2S class called ```I2SADC``` for the ESP32 port
* leverage most of the existing code (```machine_i2s_init_helper```) for initialization of this new class

Any thoughts? I really would like to see support for this and so will begin work on a PR asap.

miketeachman, 2022-04-13:
This would be a good feature to add, but I'll suggest another way to approach the API.  You could also consider extending the existing machine.ADC class.  This might be clearer for MicroPython users, as the ADC/DMA feature is related to ADC sampling, and has no relationship to the I2S protocol.   For an example on extending the machine.ADC class take a look at the [Loboris ESP32 ADC implemention](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/adc).   Loboris adds the ADC/DMA capabilities to the machine.ADC class.

oclyke, 2022-04-14:
Thanks Mike, I think that sounds like a good idea. Part of the reason I wanted to go with the route above was that the path forward seemed clear to me. I am unfamiliar with the MicroPython scheduler, and similarly unfamiliar with the methods used to make a proper asyncio/ioctl application in C. Relying on the existing I2S code was my easy way out.

Instead I'll go ahead and give it a shot to implement something quite similar in the domain of the ADC module.

oclyke, 2022-04-18:
Here's a quick update on this:
I've run into what appears to be a bug with ESP-IDF v4.2
Using the existing I2S code and adding the necessary calls to configure the ADC in I2S-driven mode I can get valid data for a short time (perhaps just one DMA buffer-full) at which point the data stops updating. I am looking into this more as I continue to develop a PR

mirko, 2023-09-21:
Hey @oclyke, thanks for your efforts!
Is there any news about that? Meanwhile mpy runs fine against ESP IDF SDK version >5.0.
Highly interested in sampling ADC values via DMA.

JohannesMaierhofer, 2023-11-20:
any news on this? I would be really waiting hard for this feature. Anything I could provide to help?

mirko, 2024-07-24:
Bump /cc @oclyke :)