micropython/micropython #8534: ESP32: calling machine.disable_irq() crashes esp

ThinkTransit, 2022-04-13:
On the esp32 port, calling machine.disable_irq() causes a crash/reboot.

Chip is ESP32-S2 and micropython version is V1.18, also happens with latest code from repo.

```
MicroPython v1.18-330-gd242a9b7f-dirty on 2022-04-13; FeatherS2 with ESP32-S2
Type "help()" for more information.
>>> import machine
>>> machine.disable_irq()
395808
>>> Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x4002f4d3  PS      : 0x00060234  A0      : 0x800302a1  A1      : 0x3ffd4460  
A2      : 0x00000001  A3      : 0x3ffc8490  A4      : 0x000003a8  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x3ffc81e8  A9      : 0x3ffc81ec  
A10     : 0x3fffca1c  A11     : 0x3ffcac1c  A12     : 0x00000000  A13     : 0x3ffcac1c  
A14     : 0x00000001  A15     : 0x3f0092b4  SAR     : 0x0000001f  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x3ffcac1c  LCOUNT  : 0x4002327c  

Backtrace:0x4002f4d0:0x3ffd4460 0x4003029e:0x3ffd4480 0x400884ba:0x3ffd44a0 0x4009fc4d:0x3ffd44c0 0x400a056f:0x3ffd44e0 0x40088280:0x3ffd4520

ELF file SHA256: 7c5cbe478bd0577b

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40023a4a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0xeb8
load:0x4004c000,len:0x888
load:0x40050000,len:0x2804
entry 0x4004c18c
MicroPython v1.18-330-gd242a9b7f-dirty on 2022-04-13; FeatherS2 with ESP32-S2
Type "help()" for more information.
>>> 
```

robert-hh, 2022-04-13:
When irq is disabled, the watchdog is not fed anymore, times out and forces a reset. So it is not a bug, it is the intended behaviour.

ThinkTransit, 2022-04-13:
Thanks Robert, I was not aware of this behaviour!

closed 2022-04-13 by ThinkTransit