micropython/micropython #8545: Enum for micropython

vovagorodok, 2022-04-14:
There is native enum implementation for micropython?

Using package of backported `enum34`:
```
>>> import enum
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400e2314  PS      : 0x00060430  A0      : 0x800e0280  A1      : 0x3ffccad0  
A2      : 0x400e09a0  A3      : 0x00000018  A4      : 0x3ffe7b14  A5      : 0x0c008136  
A6      : 0x00000407  A7      : 0x3ffe4db0  A8      : 0x00000019  A9      : 0x3ffccab0  
A10     : 0x0c008136  A11     : 0x00000000  A12     : 0x00000010  A13     : 0x3ffe8e30  
A14     : 0x00000030  A15     : 0x00000001  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0c008152  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace:0x400e2311:0x3ffccad0 0x400e027d:0x3ffccaf0 0x400e0339:0x3ffccb20 0x400e231e:0x3ffccb60 0x400e2746:0x3ffccb80 0x40083a91:0x3ffccba0 0x400dbc20:0x3ffccc40 0x400e2589:0x3ffccc70 0x400e0c22:0x3ffccc90 0x400e08fe:0x3ffcccf0 0x400e2589:0x3ffccd10 0x40084559:0x3ffccd30 0x400dbc20:0x3ffccdd0 0x400e2589:0x3ffcce40 0x400e262c:0x3ffcce60 0x400fe96d:0x3ffccef0 0x400febc6:0x3ffccf20 0x400e2734:0x3ffcd000 0x4008482d:0x3ffcd040 0x400dbc20:0x3ffcd0e0 0x400e2589:0x3ffcd130 0x400e25b2:0x3ffcd150 0x400ef916:0x3ffcd170 0x400efaa9:0x3ffcd200 0x400d4cd6:0x3ffcd250

ELF file SHA256: 0042e5dc6a30789f

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:4536
load:0x40078000,len:12344
load:0x40080400,len:4124
entry 0x40080680
MicroPython v1.18 on 2022-04-14; ESP32 module with ESP32
Type "help()" for more information.
```

Than was tried another way to `mpy-cross` CPython libs `enum.py` and `type.py`:
```
>>> import enum
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "./py/enum.py", line 3, in <module>
  File "./py/types.py", line 14, in <module>
AttributeError: 'function' object has no attribute '__code__'
```
Related topic was found: https://github.com/micropython/micropython/issues/2084
Any idea how to use emun in micropython? Is required by `python-chess` that I'm trying to use

jbbjarnason, 2022-04-19:
relates to an issue on the micropython lib repository, https://github.com/micropython/micropython-lib/issues/269

IhorNehrutsa, 2025-03-01:
[docs/library/enum.rst: Add Enum class. #16842](https://github.com/micropython/micropython/pull/16842)

Josverl, 2025-03-02:
@IhorNehrutsa , 
The (optional) MicroPython modules such as your proposal for `Enum` are managed in a separate repository : [micropython-lib](https://github.com/micropython/micropython-lib)
So apart from the documentation , the implementation of the `Enum` class will need to be in a PR to micropython-lib

IhorNehrutsa, 2025-03-05:
Implementation in:
[micropython-lib/python-stdlib/enum/enum.py: Add Enum class. #980](https://github.com/micropython/micropython-lib/pull/980)