micropython/micropython #8546: WiFi issues with Arduino Nano RP2040 Connect

peterhinch, 2022-04-14:
Ref [this forum thread](https://forum.micropython.org/viewtopic.php?f=21&t=12226) - a user was trying to use my asynchronous MQTT library without success so I investigated. It seems difficult to do much with the WiFi. My NINA-W102 module firmware is V1.4.8 which is current. Firmware build
```
MicroPython v1.18 on 2022-04-07; Arduino Nano RP2040 Connect with RP2040
```
The basic connect script produces weird results
```python
import network
sta_if = network.WLAN(network.STA_IF)
ap = network.WLAN(network.AP_IF) # create access-point interface
ap.active(False)         # deactivate the interface
if not sta_if.isconnected():
    print('connecting to network...')
    sta_if.active(True)
    sta_if.connect('NetworkName', 'Password')  # ADAPT
    while not sta_if.isconnected():
        pass
print('network config:', sta_if.ifconfig())  # Prints a pile of ordure
a = sta_if.config('mac')
print('MAC {:02x}:{:02x}:{:02x}:{:02x}:{:02x}'.format(a[0],a[1],a[2],a[3],a[4]))
```
The `ifconfig` produced garbage. Re-issuing it produced different garbage.
```python
network config: ('0.0.0.0', '8.255.1.0', '1.0.0.0', '12.0.0.0')
MAC 87:01:10:01:00
>>> print('network config:', sta_if.ifconfig())
network config: ('0.0.0.0', '189.65.0.16', '0.0.0.0', '2.0.0.0')
>>> print('network config:', sta_if.ifconfig())
network config: ('0.0.0.0', '189.65.0.16', '0.0.0.0', '2.0.0.0')
>>> 
```
I then tried to use `umqtt.simple` as follows, but it consistently failed to connect.
```python
from umqtt.simple import MQTTClient
from time import sleep

def main(server="192.168.0.10"):  # ADAPT
    c = MQTTClient("umqtt_client", server)
    c.connect()  # Fails here
    while True:
        c.publish(b"shed", b"hello", qos=1)
        sleep(5)
main()
```
These simple scripts work fine on an ESP32

robert-hh, 2022-04-14:
The culprit is the NINA Firmware, which does not provide all socket methods. See e.g. the discussion here https://github.com/micropython/micropython/pull/7668

peterhinch, 2022-04-15:
Thanks for that. Hopefully these issues will be fixed in due course because WiFi is largely unusable at the moment. There seem to be two main issues:
 1. WiFi connectivity is unstable. [#7668](https://github.com/micropython/micropython/pull/7668) seems to indicate this is getting attention.
 2. For some reason `umqtt.simple` won't connect to a local broker. This is a very basic network application.

If it's a NINA issue presumably it's also affecting Arduino users. Since this isn't a MicroPython issue I guess I should close this.

robert-hh, 2022-04-15:
> If it's a NINA issue presumably it's also affecting Arduino users.

I wondered about that as well. Adafruit uses a modified version of the NINA firmware for their Airlift product. I could install Circuitpython on the board for a test, but ... I'm a little bit lazy to deal with it.

iabdalkader, 2022-04-18:
Hi, yes the firmware is buggy and it's missing some critical sockets functions. I have Not abandoned it though, I will submit firmware fixes upstream very soon, which should fix all of these issues. That said, `ifconfig` should be working, it seems that `isconnected` is broken for some reason and always returns True. I'm going to look into it, but for now if you just ignore and connect() it should work as expected.

iabdalkader, 2022-04-18:
Okay I see the issue, `isconnected()` is called before `sta_if.active(True)` and at that point the module is Not initialized, so it returns `-1` which makes `mp_obj_new_bool` return True. This particular issue is fixed in  #8561

peterhinch, 2022-04-22:
This is now working with today's build.

closed 2022-04-22 by peterhinch