micropython/micropython #8550: rp2: Script using _thread crashes without explicit GC call

peterhinch, 2022-04-14:
This was prompted by the fact that the script in https://github.com/micropython/micropython/issues/8360 deterministically crashed at ~190K messages (with the mutex fix). The cause of the crash wasn't clear - there were no corrupt UART messages, it simply silently stopped running.

The following increases the speed by a factor of 100 and reduces code. As posted it runs indefinitely, but if the `gc.collect()` is removed it fails quickly. Failure is silent.

Two other observations.
 1. Without the delay in line 20 it fails immediately with unhelpful error messages.
 2. The script cannot be interrupted with `ctrl-c`. The only way seems to be a hardware reset or power cycle.

```python
from machine import UART
import _thread
from time import sleep_us, time
import gc

start = time()
lock = _thread.allocate_lock()
uart = UART(0, 1_000_000)

def other():
    x = 0
    while True:
        lock.acquire()
        uart.write(f"1 {x:10d}")
        lock.release()
        sleep_us(1_050)
        x += 1

_thread.start_new_thread(other, ())
sleep_us(10_000)  # Empirically necessary. Delay value is a wild guess.
n = 0
while True:
    lock.acquire()
    uart.write(f"0 {n:10d}")
    lock.release()
    sleep_us(1_000)
    n += 1
    if not n % 100:
        gc.collect()  # Without this it crashes silently
    if not n % 2000:
        print("Messages sent", n, "runtime", (time() - start)//60, "mins")
```
With the GC, it seems solid: >3M messages, both threads still running.

dpgeorge, 2022-04-15:
I tried running this test but cannot reproduce the problem.

I used a Pico board running 9ab66b50c.  I ran the above script with the "if not n % 100:" and "gc.collect()" lines commented out.  I ran it using `mpremote run --no-follow test.py repl`.  It ran for >3hrs and had >8M messages without any issue.  I was able to stop the test with ctrl-C:
```
Messages sent 8308000 runtime 186 mins
Messages sent 8310000 runtime 186 mins
[04]Traceback (most recent call last):
  File "<stdin>", line 32, in <module>
KeyboardInterrupt: 
```

robert-hh, 2022-04-15:
The test is running now for >1 hr and >2.5 Million messages without problems. "if not n % 100:" and "gc.collect()"  are commented out. I uploaded the script to the board and connected with picocom.

peterhinch, 2022-04-15:
This was evidently down to outdated firmware. All issues are now fixed with https://github.com/micropython/micropython/commit/9ab66b50cb0f15dab9c935324751edb638144099

closed 2022-04-15 by peterhinch

dpgeorge, 2022-04-18:
Maybe the commit you had did include the UART mutex fix yet still showed the issue described here, and @robert-hh and I could not reproduce on a latter commit because latter commits subtly changed the layout of heap memory?

peterhinch, 2022-04-19:
If so, that hints at a certain fragility. I'd naively imagined something had been fixed :)

I don't have a record of the exact build I was using when I started on the mutex changes but I started looking at it on 28th February. I normally update sources before making any changes.

dpgeorge, 2022-04-20:
Well, if you see it again feel free to reopen this issue.