micropython/micropython #8554: How can I get the pin_number by object of machine.Pin

MarchAwake, 2022-04-16:
import machine
machine_pin = machine.Pin(18, mode=machine.Pin.IN, pull=machine.Pin.PULL_DOWN)

# how can I use the machine_pin to get pin number 18 
# I try machine_pin.__dict__ but stderr:
# >>> machine_pin.__dict__
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'Pin' object has no attribute '__dict__'

robert-hh, 2022-04-16:
You can use e.g. str(machine_pin) to get 'Pin(18)', and int(str(machine_pin)[4:-1]) to extract the number as int.
Besides that, the place here is the issue tracker. Questions like your's are better asked in the forum at https://forum.micropython.org.

MarchAwake, 2022-04-17:
@robert-hh only passed this way to get pin number of machine.Pin instance?

robert-hh, 2022-04-17:
As far as I know, this is the only way for the ESP32 build. Othe ports may offer different options, but not necessarily better ones.

dpgeorge, 2022-04-18:
There is no standard way to get the pin number.  If you want such a feature you can try subclassing `machine.Pin`, or just create a tuple `(pin, number)` and pass that around your code.

MarchAwake, 2022-04-18:
@dpgeorge thanks a lot.