micropython/micropython #8555: machine.Pin.irq() doesn't work in a start_new_thread

MarchAwake, 2022-04-17:
import _thread
import machine

def irq_handler(machine_pin):
    pin_number = str(machine_pin)[4:6]
    print("pin number {} irq".format(pin_number))
        

def machine_pin_irq():
    print("start init irq task!")
    machine_pin_18 = machine.Pin(18, mode=machine.Pin.IN, pull=machine.Pin.PULL_DOWN)
    machine_pin_19 = machine.Pin(19, mode=machine.Pin.IN, pull=machine.Pin.PULL_DOWN)
    machine_pin_18.irq(trigger=machine.Pin.IRQ_RISING, handler=irq_handler)
    machine_pin_19.irq(trigger=machine.Pin.IRQ_RISING, handler=irq_handler)
 

if __name__ == '__main__':
    
    machine_pin_irq()                                                    # irq work well
    #_thread.start_new_thread(machine_pin_irq, ())     # it doesn't work
    print("start new thread completed")

# when I'm invoking  _thread.start_new_thread(machine_pin_irq, ()) , it doesn't work

peterhinch, 2022-04-17:
I'm not sure what you're trying to achieve here. 

The normal use of `_thread.start_new_thread()` is to cause a function to run on the other core. Typically such a function would run for a long time or forever. Running the ISR on the other core wouldn't achieve much as it (rightly) runs to completion very fast. In any event if you were to run it in this way you'd need to pass a `Pin` instance as an arg.

Support queries like this should be raised in [the forum](http://[forum](http://forum.micropython.org/index.php).micropython.org/index.php).

MarchAwake, 2022-04-18:
@peterhinch Thanks a lot.

closed 2022-04-18 by mattytrentini