micropython/micropython #8557: WiFi does not work properly on ESP32-C3-WROOM-02 (firmware: esp32c3-usb-20220117-v1.18.bin)

ellbl, 2022-04-18:
>>> import network
sta_if = network.WLAN(network.STA_IF); sta_if.active(True)
True
>>> sta_if.scan()

Connection lost (EOF)

Use Stop/Restart to reconnect.

dpgeorge, 2022-04-19:
This module doesn't use USB JTAG.  Please use the standard ESP32C3 firmware from https://micropython.org/download/esp32c3/ (I checked, it works on a ESP32-C3-WROOM-02, WiFi works).

ellbl, 2022-04-21:
Thanks for that. but I use ESP32C3-USB-20220117-V1.18.BIN firmware ON ESP32-C3-M1, WiFi works.

dpgeorge, 2022-04-21:
> I use ESP32C3-USB-20220117-V1.18.BIN firmware ON ESP32-C3-M1, WiFi works.

I'm not sure how the ESP32-C3-M1 is configured, but if the standard GENERIC_C3 firmware works for you on a ESP32-C3-WROOM-02 then there is no problem.

ellbl, 2022-04-21:
Yes. I checked, standard GENERIC_C3 firmware works  on a ESP32-C3-WROOM-02 no problem.

dpgeorge, 2022-04-21:
Then can this issue be closed?  If not please state clearly the problem.

ellbl, 2022-04-28:
Today, I tested esp32-c3-mini-1 and failed，I want to find out why esp32-c3-m1 works well？

ellbl, 2022-04-28:
Thanks for that. I hope the next version of WiFi and USB JTAG can be shared.
Now I will close this issue.

closed 2022-04-28 by ellbl