micropython/micropython #8560: How can I access non-REPL UART over USB?

sk0g, 2022-04-18:
So I'm using the RP2 port of MicroPython.

Is there a way to, at runtime, switch to EG UART1 communication? By default I get the Python REPL, which is fine but I'm trying to expose UART1, with all its relevant programming, over the USB connection. Ideally would avoid the hardware pin-connected adapter if at all possible. 

I can see USB_VCP [0] is a thing, but it's only for PyBoard. CircuitPython [1] for example allows defining another virtual USB device, and mapping it to UART from what I can tell.

[0] https://docs.micropython.org/en/latest/library/pyb.USB_VCP.html
[1] https://learn.adafruit.com/customizing-usb-devices-in-circuitpython?view=all#add-a-second-serial-port-3096573-2

stephanelsmith, 2022-04-21:
Bump on this.  I just asked yesterday over on the foums
https://forum.micropython.org/viewtopic.php?f=18&t=12293

The REPL connection to the usb stream can be problematic.  Never though about the ability to create additional virtual serial ports over USB.  That's pretty interesting.

sk0g, 2022-04-21:
@stephanelsmith there's a community sourced patch that tackles this. I don't want to rely on a shaky fix for a major limitation (IMO), but you may find it helpful still :) For clarity, adding it to the issue itself.

niutianhu, 2023-11-05:
中国上海移远通信公司移植MicroPython到了多个4G的模组，他们的模组插上usb以后会模拟出多个串口出来，不知道是不是你说的这个意思