micropython/micropython #8562: Does ussl.wrap_socket() support providing cert for client request?

PsuFan, 2022-04-18:
Does ussl.wrap_socket() support providing cert? Can someone provide an example? Don't see anything in http_client_ssl.py, using ESP8266. Also getting the urequests.py OSError -40. Even with self signed cert. Have made requests to this server with ESP8266/Arduino. 

```
import usocket
import ussl
addr = usocket.getaddrinfo('server.net', 443)[0][-1]

f = open('cert.pem')
cert = f.read()
f.close()

s = usocket.socket()
s.connect(addr)
ss = ussl.wrap_socket(s, cert=cert)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: -40

peterhinch, 2022-04-19:
If you search [the forum](http://[forum](http://forum.micropython.org/index.php).micropython.org/index.php) you will find answers to this.

jonnor, 2024-09-02:
esp8266 with SSL is unfortunately highly problematic with modern servers, as explained in https://github.com/micropython/micropython-lib/issues/400#issuecomment-2081407460
Closing as duplicate of that issue thread.

closed 2024-09-02 by jonnor

PsuFan, 2024-09-02:
It’s not an problem for the Arduino library using the same hardware…