micropython/micropython #8567: Building with 'memzip.c'

gvanem, 2022-04-20:
I'm building on Windows and toying with the `shared/memzip/*` files.
But during link, I get the error: `import.obj : error LNK2005: _mp_vfs_import_stat already defined in vfs.obj`

Not fully understanding all this *aliasing* stuff, but just patching like:
```diff
--- a/shared/memzip/import.c 2021-07-13 13:48:26
+++ b/shared/memzip/import.c 2022-04-20 15:28:29
@@ -3,6 +3,7 @@
 #include "py/lexer.h"
 #include "memzip.h"

+#undef mp_import_stat
 mp_import_stat_t mp_import_stat(const char *path) {
     MEMZIP_FILE_INFO info;
```

that link error was gone.

But I'm not sure how my generated `python-stdlib.c` is really used. How can a `MicroPython.exe -v ...` tell me it's 
importing from this file? Seems that with `-DMICROPY_VFS`, a `memzip_stat()` never gets called.

BTW, a warning:
`shared/memzip/memzip.c(64): warning C4047: 'return': 'bool' differs in levels of indirection from 'void *'`

fixed by:
```diff
--- a/shared/memzip/memzip.c 2021-07-13 13:48:26
+++ b/shared/memzip/memzip.c 2022-04-20 15:35:40
@@ -61,7 +61,7 @@
         mem_data += file_hdr->uncompressed_size;
         file_hdr = (const MEMZIP_FILE_HDR *)mem_data;
     }
-    return NULL;
+    return false;
 }
```

dpgeorge, 2022-04-21:
Memzip is not well supported on anything except the teensy port.

It might be better to explain what you're trying to achieve, rather than attempting to get memzip working.

gvanem, 2022-04-21:
> It might be better to explain what you're trying to achieve,

To use modules from a memory filesystem only. I feel such a .zip-file is also a *Virtual File System*. 
Hence a future `MICROPY_MEMZIP 1` should be set only if `MICROPY_VFS 1`. But what would it mean?
* Only access files from such a .zip-file? I.e. the one generated by `make-memzip.py`.
* Or try the .zip-file first, then the next VFS in the chain?
* And what should the default `sys.path[]` look like?

robert-hh, 2022-04-21:
Why don't you use frozen bytecode? There is no advantage using memzip, because that gets part of the firmware as well.

gvanem, 2022-04-21:
> Why don't you use frozen bytecode

How? I've no idea what you're talking about.

jonnor, 2025-10-03:
Frozen bytecode allows MicroPython to load .py that have been precompiled into .mpy and baked into the firmware image. Documentation is linked here: https://docs.micropython.org/en/latest/develop/optimizations.html#frozen-bytecode