micropython/micropython #8569: ESP32 psram - uasyncio in _thread - Core Panic

straga, 2022-04-20:
Run uasyncio in the thread.
ESP-IDF 4.4.
micropython: master

IF run not in the _thread work like before. Before a couple of months ago work right in Thread too.

```
    # AsyncIO in thread
    loop = asyncio.get_event_loop()
    _ = _thread.stack_size(8 * 1024)
    _thread.start_new_thread(loop.run_forever, ())
```

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4010a5ac  PS      : 0x00060f30  A0      : 0x800e8eb4  A1      : 0x3ffd3320
0x4010a5ac: mp_pairheap_pairing at /opt/upy/micropython/py/pairheap.c:74
 (inlined by) mp_pairheap_pairing at /opt/upy/micropython/py/pairheap.c:67

A2      : 0x400e8ebc  A3      : 0x00000000  A4      : 0x3f821590  A5      : 0x00000000
0x400e8ebc: task_lt at /opt/upy/micropython/extmod/moduasyncio.c:78

A6      : 0x00000001  A7      : 0x00000002  A8      : 0x00000000  A9      : 0x3ffd3300
A10     : 0x00000000  A11     : 0x3f821590  A12     : 0x3f821590  A13     : 0x00000008
A14     : 0x00000000  A15     : 0x00000005  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000000c  LBEG    : 0x40083a4d  LEND    : 0x40083a55  LCOUNT  : 0x00000027
0x40083a4d: esp_timer_impl_get_counter_reg at /opt/upy/tools/esp-idf-v4.4/components/esp_timer/src/esp_timer_impl_lac.c:118 (discriminator 2)

0x40083a55: esp_timer_impl_get_counter_reg at /opt/upy/tools/esp-idf-v4.4/components/esp_timer/src/esp_timer_impl_lac.c:128 (discriminator 1)

Backtrace:0x4010a5a9:0x3ffd33200x400e8eb1:0x3ffd3340 0x400dd959:0x3ffd3360 0x400e3ae5:0x3ffd3380 0x400e3c36:0x3ffd33a0 0x400860c9:0x3ffd33c0 0x400dda78:0x3ffd3460 0x400e3ae5:0x3ffd3490 0x40086019:0x3ffd34b0 0x400dda78:0x3ffd3550 0x400e3ae5:0x3ffd35a0 0x40107a8f:0x3ffd35c0 0x400d97b1:0x3ffd3660
0x4010a5a9: mp_pairheap_pairing at /opt/upy/micropython/py/pairheap.c:86

0x400e8eb1: mp_pairheap_pop at /opt/upy/micropython/py/pairheap.h:97
 (inlined by) task_queue_pop_head at /opt/upy/micropython/extmod/moduasyncio.c:127

0x400dd959: fun_builtin_1_call at /opt/upy/micropython/py/objfun.c:71

0x400e3ae5: mp_call_function_n_kw at /opt/upy/micropython/py/runtime.c:673

0x400e3c36: mp_call_method_n_kw at /opt/upy/micropython/py/runtime.c:689

0x400860c9: mp_execute_bytecode at /opt/upy/micropython/py/vm.c:1004

0x400dda78: fun_bc_call at /opt/upy/micropython/py/objfun.c:289

0x400e3ae5: mp_call_function_n_kw at /opt/upy/micropython/py/runtime.c:673

0x40086019: mp_execute_bytecode at /opt/upy/micropython/py/vm.c:919

0x400dda78: fun_bc_call at /opt/upy/micropython/py/objfun.c:289

0x400e3ae5: mp_call_function_n_kw at /opt/upy/micropython/py/runtime.c:673

0x40107a8f: thread_entry at /opt/upy/micropython/py/modthread.c:195

0x400d97b1: freertos_entry at /opt/upy/micropython/ports/esp32/build-STRAGA_CORE_SPIRAM/../mpthreadport.c:116
```

peterhinch, 2022-04-21:
See https://github.com/micropython/micropython/issues/8340.

straga, 2022-07-05:
Now all works.

closed 2022-07-05 by straga