micropython/micropython #8570: ESP32S3 UART?

TGiles1998, 2022-04-20:
Hi,
I am currently connecting a SIM800L to an ESP32 s3 using pins rx: 18, tx: 17
and running this code:

from machine import UART
id = 1
gsm = UART(id, 115200)

Result:

Guru Meditation Error: Core  1 panic'ed (LoadStoreAlignment). Exception was unhandled.

Have all three UART pairs available for the ESP32S3 been mapped?

[image]

Many thanks,

Thomas

robert-hh, 2022-04-20:
Did you try other pins, like 4 and 5 by assign tx and rx to these Pins? ESP32 usually allows free assigment of pins to functions.

stephanelsmith, 2022-04-21:
I'm using pins 17/18 on s3.  Try explictling defining the pins like this:

uart1 = UART(1, 115200, tx=17, rx=18)