micropython/micropython #8579: Ai-Thinker ESP32-S3-32S

ITstreet1, 2022-04-22:
Hi,
I have this board. According to the datasheet, it is 8MB SPI, with no PSRAM. It has UAR USB only.
https://docs.ai-thinker.com/_media/esp32/docs/nodemcu-esp-s3-32s-kit_v1.0.0specification.pdf

I successfully flashed this BIN with this command:
esptool.py --chip esp32s3 --port COM3 write_flash -z 0x0 E:GENERIC_S3-20220117-v1.18.bin

https://micropython.org/download/GENERIC_S3/

But, when I tried to work with it with Thonny, I get:

```
W (238) bootloader_random: RNG for ESP32-S3 not currently supported
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0xf60
load:0x403b6000,len:0x978
load:0x403ba000,len:0x2c80
entry 0x403b616c
W (25) bootloader_random: RNG for ESP32-S3 not currently supported
W (239) bootloader_random: RNG for ESP32-S3 not currently supported
```
It looks like this chip in the module is not supported, or I didn't choose the right BIN file.

lyzp, 2022-04-23:
Recompile a firmware, cancel usb enable

ITstreet1, 2022-04-23:
It worked only with CircuitPython 7.2.0

fo2w, 2022-05-03:
same problem

jonnor, 2025-10-03:
This board should probably work with the ESP32-S3 generic firmware on the MicroPython download page, in recent versions. Can someone test and verify?