micropython/micropython #8580: String versus bytes performance

remibert, 2022-04-23:
Hello, with the example below on esp32 (V1.18 micropython), the string is 89 times slower than the bytes, unfortunately not only on this simple case, on another esp32 I can reach a factor of 280, it depends on the hardware, a pico pi is 20 times. An optimization would save time for many applications.

**Result :**
- concatenation strings 2328 ms
- concatenation bytes 26 ms

**Source :**
from time import ticks_ms                                                                                    
start = ticks_ms()

for i in range(1000):
    a = "b" + "b"

end = ticks_ms()
print("concatenation strings %d ms"%(end-start))

start = ticks_ms()

for i in range(1000):
    a = b"b" + b"b"

end = ticks_ms()
print("concatenation bytes %d ms"%(end-start))

stephanelsmith, 2022-04-25:
That's incredible information I wouldn't have though to look into.  In my application, I can use str/bytes interchangeably.  Now I know which to prefer!

peterhinch, 2022-04-25:
The MicroPython recommendation is that using `+` to join strings is inefficient. Performance-oriented applications should use `string.join`, `string.format` or the f-string equivalent.

dpgeorge, 2022-04-25:
This will most likely be due to string interning being done on the newly-created string, which involves a search through all existing qstrs.  That doesn't happen when using bytes.

Performance will be improved by #6896.

jonnor, 2024-09-08:
The referenced MR has now been implemented, which should have improved performance a bit. Can we close this issue?

Additionally, I would say that if bytes have sufficient features for a given piece of code, it should probably be preferred over strings.

robert-hh, 2024-09-08:
I tested that on various boards, and the factor is 3 - 4.  A clear difference, but not 89.

jonnor, 2024-09-08:
That is a huge improvement! Probably close enough to consider closing this issue?

projectgus, 2025-02-04:
The difference is still big enough that some people are likely to benefit from switching to bytes objects. Have opened a docs PR for this.

PR #16691 (2025-02-11): docs: Add strings vs bytes to speed optimisation tips.
### Summary

Closes #8580. The speed difference has substantially reduced since that issue was opened, but it's still enough to make a difference in some cases and not particularly obvious.

Also add some additional context links, suggestions for alternative classes, etc.

*This work was funded through GitHub Sponsors.*
closed 2025-02-11 by dpgeorge