micropython/micropython #8582: How I can use machine modules instead pyb module in micropython to get the same answer?

IR-Consensus, 2022-04-23:
I searched in micropython source but just has methods and class for pyb module.
Example: Link (https://docs.micropython.org/en/latest/library/pyb.html).

with no support pyb module in esp32 wroom board, Can we use other modules with all the benefits?
I don't find in machine sources the needed methods and classes for Replacement.

What can I do to get the answer from the source code in ESP32-S3-WROOM?
Related methods that are not supported in source code project.
1- pyb.usb_mode().
2 - pyb.Timer()
3 - pyb.USB_HID()

Link project: https://github.com/pmvr/micropython-fido2/blob/b28616a08b2a6106df79b10e79bd59ff91999c91/boot.py 
and files hid.py - wink.py - main.py.

Please guide me.
Thanks.

jimmo, 2022-04-23:
This might be a better question for the forum. https://forum.micropython.org/

But for 1 & 3, usb_mode and USB_HID, neither of these are supported on ESP32. On the original ESP32 this is because there is no support for USB at all (most boards provide a USB UART, but this is entirely on a separate chip and UART-only). On ESP32{S2,S3,C3} we do not yet have support for the built-in USB peripheral other than UART. I think CircuitPython might though.

For 2, Instead of pyb.Timer you can use https://docs.micropython.org/en/latest/library/machine.Timer.html although it does not have quite as much functionality as pyb.Timer.

closed 2022-04-23 by jimmo

IR-Consensus, 2022-04-23:
Should the resources mentioned be Further investigated and tested for my purpose?

my mean is, it has a  management good pyb module than the support UART method in USB support?
Do I need to change the esp32 wroom board to get the answer better?

peterhinch, 2022-04-23:
The `pyb` module is specific to STM chips (e.g. Pyboard). On other platforms `machine` should be used. `pyb` has some additional functionality because this is provided by the STM hardware.

IR-Consensus, 2022-04-23:
Thank you very much.
Thank you.