micropython/micropython #8585: MicroPython ports/rp2 fails to build from source

Miki-AG, 2022-04-24:
I have followed the steps detailed in https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-python-sdk.pdf to the dot, in order to build the MicroPython ports/rp2 firmware in my system (Ubuntu 20.04.4). 
Every time I try to make ports/rp2 I get the following error:

`/home/miguel/pico/micropython/extmod/moduhashlib.c:35:10: fatal error: mbedtls/version.h: No such file or directory
   35 | #include "mbedtls/version.h"
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
/home/miguel/pico/micropython/extmod/moducryptolib.c:64:10: fatal error: mbedtls/aes.h: No such file or directory
   64 | #include <mbedtls/aes.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
`
Am I missing something here or the build is currently broken?

Full log:
`ThinkPad-E420:~/pico/micropython/ports/rp2$ make
[ -d build-PICO ] || cmake -S . -B build-PICO -DPICO_BUILD_DOCS=0 -DMICROPY_BOARD=PICO
PICO_SDK_PATH is /home/miguel/pico/micropython/lib/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
PICO target board is pico.
Using board configuration from /home/miguel/pico/micropython/lib/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: /usr/bin/python3.8 (found version "3.8.10") found components: Interpreter 
TinyUSB available at /home/miguel/pico/micropython/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
Found User C Module(s): 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/miguel/pico/micropython/ports/rp2/build-PICO
make -s -C build-PICO
Scanning dependencies of target ELF2UF2Build
[  1%] Creating directories for 'ELF2UF2Build'
[  1%] No download step for 'ELF2UF2Build'
[  2%] No patch step for 'ELF2UF2Build'
[  2%] No update step for 'ELF2UF2Build'
[  2%] Performing configure step for 'ELF2UF2Build'
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/miguel/pico/micropython/ports/rp2/build-PICO/elf2uf2
[  2%] Performing build step for 'ELF2UF2Build'
Scanning dependencies of target elf2uf2
[ 50%] Building CXX object CMakeFiles/elf2uf2.dir/main.cpp.o
[100%] Linking CXX executable elf2uf2
[100%] Built target elf2uf2
[  2%] No install step for 'ELF2UF2Build'
[  2%] Completed 'ELF2UF2Build'
[  2%] Built target ELF2UF2Build
Scanning dependencies of target bs2_default
[  3%] Building ASM object pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.obj
[  3%] Linking ASM executable bs2_default.elf
[  3%] Built target bs2_default
Scanning dependencies of target bs2_default_padded_checksummed_asm
[... 12 lines cut ...]
   64 | #include <mbedtls/aes.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
Traceback (most recent call last):
  File "/home/miguel/pico/micropython/py/makeqstrdefs.py", line 196, in <module>
    preprocess()
  File "/home/miguel/pico/micropython/py/makeqstrdefs.py", line 70, in preprocess
    for output in p.imap(pp(flags), chunks):
  File "/usr/lib/python3.8/multiprocessing/pool.py", line 868, in next
    raise value
  File "/usr/lib/python3.8/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/home/miguel/pico/micropython/py/makeqstrdefs.py", line 54, in run
    return subprocess.check_output(args.pp + flags + files)
  File "/usr/lib/python3.8/subprocess.py", line 415, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/usr/bin/arm-none-eabi-gcc', '-E', '-I/home/miguel/pico/micropython', '-I/home/miguel/pico/micropython/ports/rp2/boards/PICO', '-I/home/miguel/pico/micropython/ports/rp2', '-I/home/miguel/pico/micropython/ports/rp2/build-PICO', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_adc/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_base/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_clocks/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_dma/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_flash/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_gpio/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_i2c/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_irq/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_pio/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_pwm/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2040/hardware_regs/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_rtc/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_spi/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2040/hardware_structs/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_sync/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_timer/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_uart/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_watchdog/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/common/pico_base/include', '-I/home/miguel/pico/micropython/ports/rp2/build-PICO/generated/pico_base', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/boards/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/common/pico_binary_info/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2_common/pico_bootrom/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2_common/pico_multicore/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2_common/pico_platform/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2_common/pico_stdio/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/common/pico_stdlib/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/common/pico_sync/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/common/pico_time/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/rp2_common/pico_unique_id/include', '-I/home/miguel/pico/micropython/lib/pico-sdk/src/common/pico_util/include', '-DFFCONF_H="/home/miguel/pico/micropython/lib/oofatfs/ffconf.h"', '-DLFS1_NO_MALLOC', '-DLFS1_NO_DEBUG', '-DLFS1_NO_WARN', '-DLFS1_NO_ERROR', '-DLFS1_NO_ASSERT', '-DLFS2_NO_MALLOC', '-DLFS2_NO_DEBUG', '-DLFS2_NO_WARN', '-DLFS2_NO_ERROR', '-DLFS2_NO_ASSERT', '-DPICO_FLOAT_PROPAGATE_NANS=1', '-DPICO_STACK_SIZE=0x2000', '-DPICO_CORE1_STACK_SIZE=0', '-DPICO_PROGRAM_NAME="MicroPython"', '-DPICO_NO_PROGRAM_VERSION_STRING=1', '-DMICROPY_BUILD_TYPE="GNU 9.2.1 MinSizeRel"', '-DPICO_NO_BI_STDIO_UART=1', '-DPICO_RP2040_USB_DEVICE_ENUMERATION_FIX=1', '-DPICO_TARGET_NAME="firmware"', '-DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool', '-DMICROPY_MODULE_FROZEN_MPY=(1)', '-DPICO_BOARD="pico"', '-DLIB_PICO_MULTICORE=1', '-DLIB_PICO_PLATFORM=1', '-DPICO_NO_HARDWARE=0', '-DPICO_ON_DEVICE=1', '-DPICO_BUILD=1', '-DLIB_PICO_STDIO=1', '-DLIB_PICO_STDLIB=1', '-DLIB_PICO_SYNC=1', '-DLIB_PICO_TIME=1', '-DLIB_PICO_UNIQUE_ID=1', '-DLIB_PICO_UTIL=1', '-DLIB_TINYUSB_DEVICE=1', '-DNO_QSTR', '/home/miguel/pico/micropython/py/objtype.c', '/home/miguel/pico/micropython/py/objzip.c', '/home/miguel/pico/micropython/py/opmethods.c', '/home/miguel/pico/micropython/py/pairheap.c', '/home/miguel/pico/micropython/py/parse.c', '/home/miguel/pico/micropython/py/parsenum.c [... cut ...] moduasyncio.c', '/home/miguel/pico/micropython/extmod/modubinascii.c', '/home/miguel/pico/micropython/extmod/moducryptolib.c', '/home/miguel/pico/micropython/extmod/moductypes.c']' returned non-zero exit status 1.
make[3]: *** [CMakeFiles/firmware.dir/build.make:279: genhdr/qstr.i.last] Error 1
make[3]: *** Deleting file 'genhdr/qstr.i.last'
make[2]: *** [CMakeFiles/Makefile2:1558: CMakeFiles/firmware.dir/all] Error 2
make[1]: *** [Makefile:84: all] Error 2
make: *** [Makefile:23: all] Error 2
`

Lobo-T, 2022-04-24:
Those instructions may be a bit outdated.
Try a `make submodules` in the ports/rp2/ directory.

Miki-AG, 2022-04-24:
If I do `make submodules` followed by `make` I get the same errors:

```
user55@user55-ThinkPad-E420:~/pico/micropython/ports/rp2$ make submodules
make -f ../../py/mkrules.mk GIT_SUBMODULES="lib/mbedtls lib/pico-sdk lib/tinyusb" submodules
make[1]: Entering directory '/home/user55/pico/micropython/ports/rp2'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Updating submodules: lib/mbedtls lib/pico-sdk lib/tinyusb
Synchronizing submodule url for '../../lib/pico-sdk'
Synchronizing submodule url for '../../lib/tinyusb'
Submodule 'lib/mbedtls' (https://github.com/ARMmbed/mbedtls.git) registered for path '../../lib/mbedtls'
Cloning into '/home/user55/pico/micropython/lib/mbedtls'...
Submodule path '../../lib/mbedtls': checked out '1bc2c9cb8b8fe4659bd94b8ebba5a4c02029b7fa'
make[1]: Leaving directory '/home/user55/pico/micropython/ports/rp2'
user55@user55-ThinkPad-E420:~/pico/micropython/ports/rp2$ make
[ -d build-PICO ] || cmake -S . -B build-PICO -DPICO_BUILD_DOCS=0 -DMICROPY_BOARD=PICO
make -s -C build-PICO
[  0%] Performing build step for 'ELF2UF2Build'
[100%] Built target elf2uf2
[  0%] No install step for 'ELF2UF2Build'
[  0%] Completed 'ELF2UF2Build'
[  2%] Built target ELF2UF2Build
[  3%] Built target bs2_default
[  4%] Built target bs2_default_padded_checksummed_asm
[  4%] Generating genhdr/mpversion.h
[  5%] Generating genhdr/qstr.i.last
/home/user55/pico/micropython/extmod/moduhashlib.c:35:10: fatal error: mbedtls/version.h: No such file or directory
   35 | #include "mbedtls/version.h"
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
/home/user55/pico/micropython/extmod/moducryptolib.c:64:10: fatal error: mbedtls/aes.h: No such file or directory
   64 | #include <mbedtls/aes.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
Traceback (most recent call last):
  File "/home/user55/pico/micropython/py/makeqstrdefs.py", line 196, in <module>
    preprocess()
  File "/home/user55/pico/micropython/py/makeqstrdefs.py", line 70, in preprocess
    for output in p.imap(pp(flags), chunks):
  File "/usr/lib/python3.8/multiprocessing/pool.py", line 868, in next
    raise value
  File "/usr/lib/python3.8/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/home/user55/pico/micropython/py/makeqstrdefs.py", line 54, in run
    return subprocess.check_output(args.pp + flags + files)
  File "/usr/lib/python3.8/subprocess.py", line 415, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/usr/bin/arm-none-eabi-gcc', '-E', '-I/home/user55/pico/micropython', '-I/home/user55/pico/micropython/ports/rp2/boards/PICO', '-I/home/user55/pico/micropython/ports/rp2', '-I/home/user55/pico/micropython/ports/rp2/build-PICO', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_adc/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_base/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_clocks/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_dma/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_flash/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_gpio/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_i2c/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_irq/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_pio/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_pwm/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2040/hardware_regs/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_rtc/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_spi/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2040/hardware_structs/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_sync/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_timer/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_uart/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2_common/hardware_watchdog/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/common/pico_base/include', '-I/home/user55/pico/micropython/ports/rp2/build-PICO/generated/pico_base', '-I/home/user55/pico/micropython/lib/pico-sdk/src/boards/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/common/pico_binary_info/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2_common/pico_bootrom/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2_common/pico_multicore/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2_common/pico_platform/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2_common/pico_stdio/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/common/pico_stdlib/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/common/pico_sync/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/common/pico_time/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/rp2_common/pico_unique_id/include', '-I/home/user55/pico/micropython/lib/pico-sdk/src/common/pico_util/include', '-DFFCO [... cut ...] duasyncio.c', '/home/user55/pico/micropython/extmod/modubinascii.c', '/home/user55/pico/micropython/extmod/moducryptolib.c', '/home/user55/pico/micropython/extmod/moductypes.c']' returned non-zero exit status 1.
make[3]: *** [CMakeFiles/firmware.dir/build.make:279: genhdr/qstr.i.last] Error 1
make[3]: *** Deleting file 'genhdr/qstr.i.last'
make[2]: *** [CMakeFiles/Makefile2:1558: CMakeFiles/firmware.dir/all] Error 2
make[1]: *** [Makefile:84: all] Error 2
make: *** [Makefile:23: all] Error 2
```

Lobo-T, 2022-04-24:
Do a `make clean` or manually delete the build-PICO folder, before running make again.

Miki-AG, 2022-04-24:
@Lobo-T thanks, it worked after a 'make clean'.

closed 2022-04-24 by Miki-AG