micropython/micropython #8587: How to drive a step motor with micropython?

xky183, 2022-04-26:
Hi all,
I want drive a step motor with ESP32 and microPython,
The PWM duty will be fixed 50%, 
I want use a timer IRQ for setting the freq of the PWM with 10ms period.
The freq of PWM will be in the range of 0.5k-50kHz.
But I don't have a solution for getting the exactly number of  PWM pulse in micropython.
Is there any way to know how many pulse generated by the timer(PWM mode)? 
Thanks

xky183, 2022-04-26:
I realized that  my question is a noise for this bug tracking page and it is better  to asking in the  forum,

jimmo, 2022-04-26:
Yep, good idea to ask in the forum.

When you do, it's worth clarifying more details about what you're doing. Are you driving an h-bridge directly, or using a dedicated step/direction controller. ESP32 also has an MCPWM peripheral which might be of interest to you (see https://github.com/micropython/micropython/pull/5818).

closed 2022-04-26 by jimmo