micropython/micropython #8594: Enable factory reset boot mode without a user switch

cdwilson, 2022-04-29:
The documentation at https://docs.micropython.org/en/latest/pyboard/general.html describes a "filesystem reset" boot mode which can be triggered by holding down the user switch during the power up boot sequence:

> filesystem reset: resets the flash filesystem to its factory state, then boots in safe mode.

For supported MicroPython boards that do not have a user switch (e.g. [SparkFun MicroMod STM32 Processor](https://www.sparkfun.com/products/17713)), is there any way to trigger a "filesystem reset" from software?

If not, would it be possible to add a new `machine.factory_reset()` function to enable this?  (similar to the `machine.reset()` and `machine.soft_reset()` functions described in https://docs.micropython.org/en/latest/library/machine.html?highlight=reset#machine.reset)

For the sake of completeness, it would also be nice to support resetting into the "safe boot" boot mode from software as well through something like a `machine.safe_reset()` function.

jonnor, 2024-09-08:
`machine.bootloader()` now allows to enter the bootloader on most ports. I believe this fixes the issue?

cdwilson, 2024-09-17:
@jonnor the original request above was twofold:

1. A function to [factory reset the filesystem](https://docs.micropython.org/en/latest/pyboard/tutorial/reset.html#factory-reset-the-filesystem) (e.g. `machine.factory_reset()`)
2. A function to [reboot in safe mode](https://docs.micropython.org/en/latest/pyboard/tutorial/reset.html#safe-mode) (e.g. `machine.safe_reset()`)

As far as I can tell, these are different from `machine.bootloader()` and not currently possible to achieve without a button/switch on the board. This issue is requesting a way to trigger these from software without a button/switch.

Can we remove the `proposed-close` label until these are either implemented or rejected?

jonnor, 2024-09-29:
Fair point, thank you :) Label removed