micropython/micropython #8596: Unix port: `uasyncio` does not have certain classes/methods

GimmickNG, 2022-04-29:
The `uasyncio` library does not have the `ThreadSafeFlag`, `Lock` and possibly other classes (I haven't checked the rest), and the `gather` method does not work either. I get the following errors when running on release 1.18:

```
MicroPython v1.18 on 2022-04-29; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import uasyncio as asyncio
>>> asyncio.ThreadSafeFlag
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "uasyncio/__init__.py", line 26, in __getattr__
AttributeError: ThreadSafeFlag
>>> asyncio.gather
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "uasyncio/__init__.py", line 27, in __getattr__
ImportError: no module named 'uasyncio.funcs'
>>> asyncio.Lock
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "uasyncio/__init__.py", line 27, in __getattr__
ImportError: no module named 'uasyncio.lock'
>>> asyncio.__version__
(3, 0, 0)
```

I'm not sure whether this is related, but when creating the Unix port with the release zip, I couldn't run `make submodules` as it wasn't a git directory. When I tried to build from the master branch, I couldn't import `uasyncio` as it said the .mpy versions were incompatible.

dlech, 2022-04-29:
https://github.com/micropython/micropython/issues/6109#issuecomment-1107057608 is probably relevant

GimmickNG, 2022-05-01:
> [#6109 (comment)](https://github.com/micropython/micropython/issues/6109#issuecomment-1107057608) is probably relevant

Thanks, that worked!

closed 2022-05-01 by GimmickNG