micropython/micropython #8597: ESP8266: Purpose of / bug in WDT.deinit

czietz, 2022-04-29:
The ESP8266 has two separate watchdog timers:

- The so-called "software watchdog", which isn't 100% software, as it utilizes a timer and the corresponding interrupt routine. It has a timeout of 2 - 3 seconds, it is documented in the Espressif's SDK and can be stopped with the `system_soft_wdt_stop()` API call.
- A true hardware watchdog. This is vaguely referenced in the SDK documentation, but only [documented unofficially by third parties](https://mongoose-os.com/blog/esp8266-watchdog-timer/). In particular, there is an undocumented `ets_wdt_disable()` function in ROM. This watchdog has a timeout of 6 - 8 seconds.

Actually, _both_ watchdogs are fed by the API call `system_soft_wdt_feed()`, which is called from the appropriate functions in Micropython, e.g.
https://github.com/micropython/micropython/blob/52add7b45c944f5302e2255a848f4c0d3703587b/ports/esp8266/ets_alt_task.c#L149

What I don't understand, though, is the implementation of `machine.WDT` for the ESP8266. When a new WDT object is created, Micropython stops the automatic feeding of the watchdogs (by setting `ets_loop_dont_feed_sw_wdt`):
https://github.com/micropython/micropython/blob/52add7b45c944f5302e2255a848f4c0d3703587b/ports/esp8266/machine_wdt.c#L50-L54
... which is understandable, as it is now the users responsibility to feed them via the `feed()` method:
https://github.com/micropython/micropython/blob/52add7b45c944f5302e2255a848f4c0d3703587b/ports/esp8266/machine_wdt.c#L60-L64 
... that then calls `system_soft_wdt_feed()`.

However, there is also an `deinit()` method whose purpose I don't understand:
https://github.com/micropython/micropython/blob/52add7b45c944f5302e2255a848f4c0d3703587b/ports/esp8266/machine_wdt.c#L67-L71

It stops the _hardware_ watchdog but keeps the _software_ watchdog running. It doesn't re-enable automatic feeding, either. Therefore, ca. 2 seconds after running this minimal example, the ESP8266 will go through reset due to the software watchdog expiration:
```python
import machine

w = machine.WDT()
w.feed()
w.deinit()
```

This leads to my question: **What is the intended purpose of `deinit()`?**

- If it is to free the user again of the responsibility to call `feed()`, it should re-enable automatic feeding (by resetting `ets_loop_dont_feed_sw_wdt`).
- If, in contrast, it is meant to completely stop the watchdog (so it does not need to be fed, anymore), it has to disable not only the hardware but also the software watchdog.

czietz, 2022-05-04:
Based on @dpgeorge's very clear statement regarding watchdogs in https://github.com/micropython/micropython/issues/8600#issuecomment-1116773752
> The watchdog is not intended to be stopped. That is a feature.

... I now question why there even is a `deinit()` method for the ESP8266 watchdog. It should be removed.

dpgeorge, 2022-05-19:
The deinit method was removed in ea1c0557e5866ad7b84f18de18d0bf82e422e279

closed 2022-05-19 by dpgeorge