micropython/micropython #8600: How can I turn off the WDT in RP2

MarchAwake, 2022-04-30:
print("ENABLE WATCH DOG MONITOR!")
    wdt = machine.WDT(
        =5000)
    while True:
        wdt.feed()
        await uasyncio.sleep(1)

czietz, 2022-04-30:
The Raspberry Pi Pico SDK unfortunately only exposes an API to _enable_ the watchdog, but none to disable it again:
https://github.com/raspberrypi/pico-sdk/blob/2062372d203b372849d573f252cf7c6dc2800c0a/src/rp2_common/hardware_watchdog/include/hardware/watchdog.h
(If you think this is an oversight, you could open an issue in their repository.)

Therefore, Micropython doesn't give you a way to stop the WDT, either. There is one really, really, really ugly (and of course non-portable) way, though: directly resetting the respective bit in the watchdog control register.
```python
# Stop/disable the RP2040 watchdog timer
# 0x40058000 = WATCHDOG_CTRL register, bit 30 is the ENABLE bit
machine.mem32[0x40058000] = machine.mem32[0x40058000] & ~(1<<30)
```
This bit is officially documented in the RP2040 datasheet.

MarchAwake, 2022-05-02:
thanks a lot @czietz

dpgeorge, 2022-05-04:
The watchdog is not intended to be stopped.  That is a feature.

closed 2022-05-04 by dpgeorge

jsilversto, 2022-12-13:
Since the max WDT timeout is ~8 s, applications which have components which take any longer than this cannot use the WDT if it cannot be disabled. This is standard functionality and should be implemented.

peterhinch, 2022-12-14:
This is a FAQ and would be better raised in discussions. The purpose of a WDT is to reset the system if the code fails in any way. If a disable method exists, the possibility exists that failing code might call the method. Consequently a well designed WDT has hardware which precludes any means of cancelling it.

foxt, 2023-06-30:
@peterhinch This logic fails when you, the developer want to break out of your program.

If your code immediately sets a WDT, if you try and break into REPL under MicroPython, it will reboot.3 This is especially a problem if you set the WDT low

peterhinch, 2023-07-02:
You design a means of not enabling the WDT. You might provide for a wire link on your PCB. If you boot with the wire link in place, your code doesn't use the WDT. There are numerous other (obvious) ways of achieving the same thing. This is all standard industry practice. What you emphatically don't have is a disable method, for the reason stated.

MilhouseVH, 2025-12-10:
Just an update to mention that for the RP2350 the `WATCHDOG_BASE` register has moved to `0x400d8000`.

For the RP2040, it is at `0x40058000`.