micropython/micropython #8602: Possible port to ANSI-C?

c0pperdragon, 2022-05-01:
I would love to port micropython to my OS816 board. Main problem is that there is no C99 compiler available for the 65C816 processor. The compiler provided by the manufacturer only supports ANSI-C (C89).
Are the C99 language features really cruical for the interpreter core implementation? If no, how difficult would it be to change it to use C89 language features only?

jimmo, 2022-05-01:
I suspect this is going to be difficult unfortunately...

Here's three (just first things off the top of my head) examples of features that C89 doesn't support that we use.
 - `inline` -- in theory this isn't required (e.g. we already #define it out on Windows)
 - Single-line (C++ style) comments. This is easy to remove by preprocessing (or grep/xargs/sed). (Some compilers do have an extension for allowing this though).
 - for loop initial declarations (e.g. `for(int i = 0; ...`).

The second two aren't likely to be things that this project changes.

One possible option might be to wrap `CC` to use a tool to convert the code on the host before compiling for the target (possibly after first running it though a C99 pre-processor). I found https://github.com/libav/c99-to-c89/ It hasn't been updated for a while.

c0pperdragon, 2022-05-01:
While the compiler actually supports the // style comments, there is no way to make it understand the more flexible ways to declare local variables (inside for loops, or on any other place than the begin of a block). As this is probably a very important part of your coding style, changing this is no option.
A converter tool may indeed be the better option here. Thanks for the link, I will give it a try.

But there is one feature of the C99 that can surely not be translated to C89, which are stackframes of variable size. By which I mean that you can declare a local array which has a size determined at runtime. Also the 65C816 compiler can only support individual stackframes with less than 256 bytes (a consequence of the addressing modes of this CPU). Could this already be a showstopper?

dpgeorge, 2022-05-02:
> But there is one feature of the C99 that can surely not be translated to C89, which are stackframes of variable size. By which I mean that you can declare a local array which has a size determined at runtime.

MicroPython does not require `alloca()`.  See `MICROPY_NO_ALLOCA` option.  See also `MICROPY_ENABLE_PYSTACK`

closed 2022-12-13 by c0pperdragon