micropython/micropython #8603: Closures and nested scopes/classes.

arizvisa, 2022-05-02:
Heyo, guys. First of all I want to thank you for stewarding this project. 

I couldn't find anywhere in the documentation if whether or not nested scopes (with regards to closures and classes that are closures) are intended to be supported within µ-python. So if they're not supposed to be, then this issue can totally be closed and the rest of this issue can be ignored.

While testing the following code (ports/unix v1.18, and then 52add7b45c944f5302e2255a848f4c0d3703587b), I noticed that when the name of a closure has the same name as a method, it seems that the method does not actually get defined or is rather inaccesible (raises AttributeError)
```
def test_scope():
    def method(*self):           # [1] name of closure
        return True
    class A:
        def method(*self):       # [2] name of method
            return True

    if not hasattr(A, 'method'):
        print('nope')                # planting a dir(A) gives ['__class__', '__module__', '__name__', '__qualname__', '__bases__', '__dict__']

    x = A()
    if not hasattr(x, 'method'):
        print('nope again')          # planting a dir(x) gives ['__class__', '__module__', '__qualname__', '__dict__']
```
However, if we change the name of closure [1], to something other than the method name "method" (referenced at [2]), then things work as intended. In the following log (of the prior-mentioned test-case, I cleaned it up a bit and prefix'd the lines containing my notes.
```
>>> dispatching to test_scope()
load name test_scope
load global test_scope
calling function 7faade1b6280(n_args=0, n_kw=0, args=7ffd116138a0)
Input n_args: 0, n_kw: 0
Input pos args: 7ffd116138a0:
Input kw args: 7ffd116138a0:
Calling: n_pos_args=0, n_kwonly_args=0
7ffd11613708:
7ffd116136d8: 0 0 0 7faade1b62a0 0 0

make_function_from_raw_code 7faade1b6200
load_build_class
make_closure_from_raw_code 7faade1b6600 1 7ffd116136e0
make_function_from_raw_code 7faade1b6600
calling function 45b4a0(n_args=2, n_kw=0, args=7ffd116136e0)
calling function 7faade1b6380(n_args=0, n_kw=0, args=0)
calling function 7faade1b6360(n_args=1, n_kw=0, args=7ffd11613450)
Input n_args: 1, n_kw: 0
Input pos args: 7ffd11613450: 7faade1b62a0
Input kw args: 7ffd11613458:
Calling: n_pos_args=1, n_kwonly_args=0
7ffd116133b0: 7faade1b62a0
7ffd116133a8: 0 7faade1b62a0

load name __name__
load global __name__
store name __module__ <- aa
mp_map_rehash(7faade1b63a8): 0 -> 2
store name __qualname__ <- 17ba
make_function_from_raw_code 7faade1b6800
calling function 459ae0(n_args=3, n_kw=0, args=7ffd11613500)

load global hasattr
calling function 45b240(n_args=2, n_kw=0, args=7ffd116136e0)
mp_obj_class_lookup: Looking up method in A
mp_obj_class_lookup: No more parents

load global print
calling function 45b310(n_args=1, n_kw=0, args=7ffd116136e0)
>>> "nope"

calling function 7faade1b6480(n_args=0, n_kw=0, args=7ffd116136e0)
mp_obj_class_lookup: Looking up __new__ in A
mp_obj_class_lookup: No more parents
mp_obj_class_lookup: Looking up __init__ in A
mp_obj_class_lookup: No more parents

load global hasattr
calling function 45b240(n_args=2, n_kw=0, args=7ffd116136e0)
mp_obj_class_lookup: Looking up method in A
mp_obj_class_lookup: No more parents
mp_obj_class_lookup: Looking up __getattr__ in A
mp_obj_class_lookup: No more parents

load global print
calling function 45b310(n_args=1, n_kw=0, args=7ffd116136e0)
>>> "nope again"
```

This seems to read to me that the 3 functions (operations?) that do this are: `load_build_class`, `make_closure_from_raw_code`, and `make_function_from_raw_code`. This seems to imply that the class definition itself is not being completely parsed during `load_build_class` in certain instances or really the class itself is not being treated as a separate closure.

After figuring out that I can enable logs, I wrote another test-case to confirm my suspicions and remove out all "callables" entirely. As a result, the following sample emits only `load_build_class` and `make_function_from_raw_code`.
```
def test_scope():
    capture_1 = 5
    class closure_class:
        capture_1 = 10
    return closure_class().capture_1
```
(similar to my first testcase, this raises `AttributeError: 'closure_class' object has no attribute 'capture_1'`)

In the log for that sample, it should be storing "capture_1"
```
store name __module__ <- aa
mp_map_rehash(7fc3210d8388): 0 -> 2
malloc 32 : 7fc3210d83a0
free 0, 0
store name __qualname__ <- 17ba
calling function 459ae0(n_args=3, n_kw=0, args=7ffdf584b8e0)
```

However if we change the test-case to the following (which renames the `closure_class.capture_1` variable to `closure_class.capture_2`)
```
def test_scope():
    capture_1 = 5
    class closure_class:
        capture_2 = 10
    return closure_class().capture_2
```

Then the following log (of the just recently listed sample) show's it succeeding.
```
load name test_scope                                                            
load global test_scope                                                          
calling function 7f58d2a58280(n_args=0, n_kw=0, args=7ffc9a831520)
...
load_build_class                                                                
make_function_from_raw_code 7f58d2a58200      
...
store name __module__ <- aa
mp_map_rehash(7f58d2a58348): 0 -> 2
malloc 32 : 7f58d2a58360
free 0, 0
store name __qualname__ <- 17ba
store name capture_2 <- 15                              # here's capture_2
mp_map_rehash(7f58d2a58348): 2 -> 4
malloc 64 : 7f58d2a58380
free 7f58d2a58360, 32
calling function 459ae0(n_args=3, n_kw=0, args=7ffc9a831180)
...
calling function 459ae0(n_args=3, n_kw=0, args=7ffc9a831180)
malloc 120 : 7f58d2a583c0
calling function 7f58d2a583c0(n_args=0, n_kw=0, args=7ffc9a831360)
mp_obj_class_lookup: Looking up __new__ in closure_class
mp_obj_class_lookup: No more parents
malloc 32 : 7f58d2a58360
mp_obj_class_lookup: Looking up __init__ in closure_class
mp_obj_class_lookup: No more parents
load attr 7f58d2a58360.capture_2
load method 7f58d2a58360.capture_2
mp_obj_class_lookup: Looking up capture_2 in closure_class
mp_obj_class_lookup: Returning: 10
store name x <- 15
```

Sorry about the logs and formatting, I can clean them up if that makes it any easier. :-/

dpgeorge, 2022-05-02:
Thanks for the detailed report!

> I couldn't find anywhere in the documentation if whether or not nested scopes (with regards to closures and classes that are closures) are intended to be supported within µ-python.

Yes they are supported, and should match CPython semantics.

You have found a very interesting bug in how closures are compiled.  A reduced test case for your first example is:
```python
def test_scope():
    def method():
        pass

    class A:
        def method():
            pass

    print(hasattr(A, 'method'))
    print(hasattr(A(), 'method'))

test_scope()
```
It should print true twice, but it prints false twice.

The simpler version with just attributes `capture_1` shows the same problem.

> the class itself is not being treated as a separate closure.

This is close, but it's not as simple as this.  Class bodies apparently need to be considered a closure because of the following test:
```python
def test_scope(x):
    def closure():
        return x

    class A:
        def method2():
            return closure()

    closure = lambda: x + 1  # change it after A has been created!
    return A

print(test_scope(42).method2())
```
That prints 43.

It's not clear exactly what the semantics are for class bodies, how they should handle closed over variables.  Because if you explicitly declare `method` to be non-local then CPython behaves like MicroPython:
```python
def test_scope():
    def method():
        return 1

    class A:
        nonlocal method
        def method():
            return 2

    print(hasattr(A, "method"))
    print(hasattr(A(), "method"))
    return method()  # this is actually A.method!

print(test_scope())  # prints 2
```

arizvisa, 2022-05-02:
> This is close, but it's not as simple as this. ...

Ah hah, tru. So essentially class bodies or anything that may contain a function/method/whatever "definition" (using this term loosely) must also be treated as a closure. That's definitely \_only\_ functions and classes, right?

I ended up encountering this issue specifically when adding a couple of unit-tests earlier on Sunday where all my tests use closures heavily (in both functions/classmethods/methods/properties, and classes themselves).

> It's not clear exactly what the semantics are for class bodies, how they should handle closed over variables. Because if you explicitly declare `method` to be non-local then CPython behaves like MicroPython:

Ah, very interesting. I try to avoid some of the additional Py3-specific syntaxes in order to retain compatibility with Py2. But in this `nonlocal` sample you provided, it's capturing the "method" variable from the function it's defined in, and reassigns it with the method from the class definition. I can't quite put my finger on it, but that's almost weird semantics because in Py2, I don't think it's possible to overwrite the scope with a callable in that similar way.

So it's like during the introduction of `nonlocal`, it sorta gives you the side effect of being able modify the entire reference of another variable in a captured scope without doing an "explicit" assignment (PEP-20, Zen of Python). I don't think it like entirely allows a closure to interfere with it's captured scope if it gets the names right.

In order to get nonlocal semantics to remain-compatible with Py2, I'm using super-bad™ tricks to copy the `cell` from a closure, and re-constructing them in another code type so that I can inject references into the scope of the closure, heh.

Anyways, thanks for the insights. If there's anything I can do to help troubleshoot this further, let me know. This is sort of a blocker for me, but it's a weekend project I'm doing for a friend so there's no rush or anything.

dpgeorge, 2022-05-03:
Luckily the fix looks straightforward: #8612

arizvisa, 2022-05-03:
Haha. Awesome. You're the man. :)

arizvisa, 2022-05-03:
And that PR seems to allow my unit-tests to work. Hella thanks.

dpgeorge, 2022-05-03:
Actually the CI is failing... it's more complicated that I thought.

dpgeorge, 2022-05-03:
Fixed by 590de399f01cd08aa4825b26b91785e07abcf68c

closed 2022-05-03 by dpgeorge