micropython/micropython #8606: Spaces in f-strings can cause incorrect SyntaxErrors

mattytrentini, 2022-05-02:
Here are two examples of valid f-strings that currently raise `SyntaxErrors`:

```python
>>> x = 10
>>> f"{x = }"
SyntaxError: invalid syntax
>>> f"{x = :.3f}"
SyntaxError: invalid syntax
```

Removing the spaces between the curlies is the recommended workaround.

(The examples were found when browsing [Python f-strings Are More Powerful Than You Might Think](https://towardsdatascience.com/python-f-strings-are-more-powerful-than-you-might-think-8271d3efbd7d))

These issues should either be documented (in [cpydiff](https://github.com/micropython/micropython/tree/master/tests/cpydiff)) or resolved. 

To resolve, see the comment in https://github.com/micropython/micropython/commit/692d36d779192f32371f7f9daa845b566f26968d for background. A solution may be possible by modifying the lexer to strip spaces from the f-string before translating it to `str.format`.

jimmo, 2022-05-02:
Thanks @mattytrentini 

For reference, CPython says: "When the equal sign '=' is provided, the output will have the expression text, the '=' and the evaluated value. Spaces after the opening brace '{', within the expression and after the '=' are all retained in the output."

We probably should fix this because the SyntaxError is confusing. Like you suggest, stripping the trailing spaces from the format argument but keeping them in the format string does do the trick. First attempt below...

```diff
diff --git a/py/lexer.c b/py/lexer.c
index ac406bd46..4df829904 100644
--- a/py/lexer.c
+++ b/py/lexer.c
@@ -375,10 +375,14 @@ STATIC void parse_string_literal(mp_lexer_t *lex, bool is_raw, bool is_fstring)
                         vstr_add_byte(&lex->fstring_args, c);
                         next_char(lex);
                     }
+                    size_t end = lex->fstring_args.len;
+                    while (unichar_isspace(lex->fstring_args.buf[lex->fstring_args.len - 1])) {
+                        lex->fstring_args.len--;
+                    }
                     if (lex->fstring_args.buf[lex->fstring_args.len - 1] == '=') {
                         // if the last character of the arg was '=', then inject "arg=" before the '{'.
                         // f'{a=}' --> 'a={}'.format(a)
-                        vstr_add_strn(&lex->vstr, lex->fstring_args.buf + i, lex->fstring_args.len - i);
+                        vstr_add_strn(&lex->vstr, lex->fstring_args.buf + i, end - i);
                         // remove the trailing '='
                         lex->fstring_args.len--;
                     }
```

That's ~184 bytes though, might need some thought.

holdenweb, 2022-05-07:
I thought we aimed at 3.4 compatibility. This is from a much later release (3.8), so think yourselves lucky you got it at all! The Python documentation says "To display both the expression text and its value after evaluation, (useful in debugging), an equal sign '=' may be added after the expression." It doesn't say anything about spaces!

jimmo, 2022-05-08:
> I thought we aimed at 3.4 compatibility.
3.4 is kind of the baseline, however when features get broad usage in real Python code it can be very useful to support them in MicroPython so that existing Python code "just works". f-strings are a good example of this.

> This is from a much later release (3.8)
Yeah, f-strings are from 3.6, so it's a feature we're starting to see a lot of, but like you say the `=` modifier is from 3.8.

>  It doesn't say anything about spaces.
It does -- the paragraph I quoted above about retaining spaces is a few lines below the bit you quoted.

I do think it would be good to fix this... the SyntaxError is surprising and not necessarily obvious how to fix to someone who doesn't know the ins-and-outs of f-strings. Ideally it would raise NotImplemented, but that would be as much (or more) code than just making it work :)