micropython/micropython #8607: Micropython ESP-IDF V5.0 "Failed to resolve component 'tcpip_adapter'"

TGiles1998, 2022-05-02:
Hi,

Just wondering how this might be resolved? https://github.com/espressif/esp-idf/issues/8883#issue-1223422255

-- Check for working CXX compiler: /Users/thomasgiles/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at /Users/thomasgiles/esp/esp-idf/tools/cmake/build.cmake:197 (message):
Failed to resolve component 'tcpip_adapter'.
Call Stack (most recent call first):
/Users/thomasgiles/esp/esp-idf/tools/cmake/build.cmake:225 (__build_resolve_and_add_req)
/Users/thomasgiles/esp/esp-idf/tools/cmake/build.cmake:512 (__build_expand_requirements)
/Users/thomasgiles/esp/esp-idf/tools/cmake/project.cmake:422 (idf_build_process)
CMakeLists.txt:58 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/thomasgiles/XX/micropython/ports/esp32/build-GENERIC/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
make: *** [all] Error 2

Many thanks,

Thomas

kindmartin, 2022-05-13:
It happens the same to me. any hint?

martin@ubuntu:~/esp/esp-idf/micropython/ports/esp32$ make j=16 BOARD=GENERIC_SPIRAM
idf.py -D MICROPY_BOARD=GENERIC_SPIRAM -B build-GENERIC_SPIRAM  build
Executing action: all (aliases: build)
Running cmake in directory /home/martin/esp/esp-idf/micropython/ports/esp32/build-GENERIC_SPIRAM
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DMICROPY_BOARD=GENERIC_SPIRAM -DCCACHE_ENABLE=0 /home/martin/esp/esp-idf/micropython/ports/esp32"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/bin/git (found version "2.30.2") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/martin/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/martin/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/martin/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at /home/martin/esp/esp-idf/tools/cmake/build.cmake:197 (message):
  Failed to resolve component 'tcpip_adapter'.
Call Stack (most recent call first):
  /home/martin/esp/esp-idf/tools/cmake/build.cmake:225 (__build_resolve_and_add_req)
  /home/martin/esp/esp-idf/tools/cmake/build.cmake:512 (__build_expand_requirements)
  /home/martin/esp/esp-idf/tools/cmake/project.cmake:422 (idf_build_process)
  CMakeLists.txt:58 (project)

-- Configuring incomplete, errors occurred!
See also "/home/martin/esp/esp-idf/micropython/ports/esp32/build-GENERIC_SPIRAM/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
make: *** [Makefile:34: all] Error 2
martin@ubuntu:~/esp/esp-idf/micropython/ports/esp32$ ^C

jimmo, 2022-05-13:
IDF v5.x is not currently supported.

As Ivan said on https://github.com/espressif/esp-idf/issues/8883#issue-1223422255, MicroPython will need to migrate to using [ESP-NETIF](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_netif.html) as per the instructions at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/tcpip-adapter.html

If anyone is willing to take on that work that would be great.

We will need to come up with a way to conditionally use both ways depending on IDF version as are likely to need to support IDF 4.x for a while. We did this for some 3.x to 4.x migration too.

closed 2022-05-13 by jimmo