micropython/micropython #8615: Integrating MicroPython as a ESP-IDF component

huming2207, 2022-05-03:
Hi there,

I'm trying to integrate MicroPython into some of our existing ESP32 firmware projects, as an ESP-IDF component instead of making it as a standalone ESP32 firmware. 

Here's my repo: 
- the wrapper ESP-IDF component: https://github.com/huming2207/esp-idf-mpy
- the main demo project: https://github.com/huming2207/esp-mpy-demo

So far it seems like something went wrong when the build system generates the QSTR headers. Here's a part of the error message:

```
 jackson@jackson-desktop  ~/Projects/esp-mpy-demo   master ±  idf.py build    
Executing action: all (aliases: build)
Running ninja in directory /home/jackson/Projects/esp-mpy-demo/build
Executing "ninja all"...
[3/104] Building C object esp-idf/mpy/CMakeFiles/__idf_mpy.dir/mpy/py/modcollections.c.obj
FAILED: esp-idf/mpy/CMakeFiles/__idf_mpy.dir/mpy/py/modcollections.c.obj 
/home/jackson/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -DFFCONF_H=\"/ffconf.h\" -DLFS1_NO_ASSERT -DLFS1_NO_DEBUG -DLFS1_NO_ERROR -DLFS1_NO_MALLOC -DLFS1_NO_WARN -DLFS2_NO_ASSERT -DLFS2_NO_DEBUG -DLFS2_NO_ERROR -DLFS2_NO_MALLOC -DLFS2_NO_WARN -DMICROPY_ESP_IDF_4=1 -DMICROPY_VFS_FAT=1 -DMICROPY_VFS_LFS2=1 -I/home/jackson/Projects/esp-mpy-demo/build/config -I/home/jackson/Projects/esp-mpy-demo/components/mpy/include -I/home/jackson/Projects/esp-mpy-demo/components/mpy/mpy -I/home/jackson/Projects/esp-mpy-demo/components/mpy/mpy/shared/runtime -I/home/jackson/Projects/esp-mpy-demo/build -I/home/jackson/esp/esp-idf/components/newlib/platform_include -I/home/jackson/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/jackson/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/home/jackson/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/jackson/esp/esp-idf/components/freertos/esp_additions/include -I/home/jackson/esp/esp-idf/components/esp_hw_support/include -I/home/jackson/esp/esp-idf/components/esp_hw_support/include/soc -I/home/jackson/esp/esp-idf/components/esp_hw_support/include/soc/esp32s3 -I/home/jackson/esp/esp-idf/components/esp_hw_support/port/esp32s3/. -I/home/jackson/esp/esp-idf/components/esp_hw_support/port/esp32s3/private_include -I/home/jackson/esp/esp-idf/components/heap/include -I/home/jackson/esp/esp-idf/components/log/include -I/home/jackson/esp/esp-idf/components/lwip/include/apps -I/home/jackson/esp/esp-idf/components/lwip/include/apps/sntp -I/home/jackson/esp/esp-idf/components/lwip/lwip/src/include -I/home/jackson/esp/esp-idf/components/lwip/port/esp32/include -I/home/jackson/esp/esp-idf/components/lwip/port/esp32/include/arch -I/home/jackson/esp/esp-idf/components/soc/include -I/home/jackson/esp/esp-idf/components/soc/esp32s3/. -I/home/jackson/esp/esp-idf/components/soc/esp32s3/include -I/home/jackson/esp/esp-idf/components/hal/esp32s3/include -I/home/jackson/esp/esp-idf/components/hal/include -I/home/jackson/esp/esp-idf/components/hal/platform_port/include -I/home/jackson/esp/esp-idf/components/esp_rom/include -I/home/jackson/esp/esp-idf/components/esp_rom/include/esp32s3 -I/home/jackson/esp/esp-idf/components/esp_rom/esp32s3 -I/home/jackson/esp/esp-idf/components/esp_common/include -I/home/jackson/esp/esp-idf/components/esp_system/include -I/home/jackson/esp/esp-idf/components/esp_system/port/soc -I/home/jackson/esp/esp-idf/components/esp_system/port/include/private -I/home/jackson/esp/esp-idf/components/xtensa/include -I/home/jackson/esp/esp-idf/components/xtensa/esp32s3/include -I/home/jackson/esp/esp-idf/components/vfs/include -I/home/jackson/esp/esp-idf/components/esp_wifi/include -I/home/jackson/esp/esp-idf/components/esp_event/include -I/home/jackson/esp/esp-idf/components/esp_netif/include -I/home/jackson/esp/esp-idf/components/esp_eth/include -I/home/jackson/esp/esp-idf/components/esp_phy/include -I/home/jackson/esp/esp-idf/components/esp_phy/esp32s3/include -I/home/jackson/esp/esp-idf/components/bootloader_support/include -I/home/jackson/esp/esp-idf/components/bootloader_support/bootloader_flash/include -I/home/jackson/esp/esp-idf/components/esp_timer/include -mlongcalls  -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fmacro-prefix-map=/home/jackson/Projects/esp-mpy-demo=. -fmacro-prefix-map=/home/jackson/esp/esp-idf=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.0-dev-2586-ga82e6e63d9\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/mpy/CMakeFiles/__idf_mpy.dir/mpy/py/modcollections.c.obj -MF esp-idf/mpy/CMakeFiles/__idf_mpy.dir/mpy/py/modcollections.c.obj.d -o esp-idf/mpy/CMakeFiles/__idf_mpy.dir/mpy/py/modcollections.c.obj -c /home/jackson/Projects/esp-mpy-demo/components/mpy/mpy/py/modcollections.c
In file included from /home/jackson/Projects/esp-mpy-demo/components/mpy/mpy/py/mpconfig.h:62,
                 from /home/jackson/Projects/esp-mpy-demo/components/mpy/mpy/py/obj.h:31,
                 from /home/jackson/Projects/esp-mpy-demo/components/mpy/mpy/py/builtin.h:29,
                 from /home/jackson/Projects/esp-mpy-demo/components/mpy/mpy/py/modcollections.c:27:
/home/jackson/Projects/esp-mpy-demo/components/mpy/include/mpconfigport.h:66:36: warning: backslash-newline at end of file
 #define MICROPY_PORT_ROOT_POINTERS \
                                     
In file included from /home/jackson/Projects/esp-mpy-demo/components/mpy/mpy/py/builtin.h:29,
                 from /home/jackson/Projects/esp-mpy-demo/components/mpy/mpy/py/modcollections.c:27:
/home/jackson/Projects/esp-mpy-demo/components/mpy/mpy/py/modcollections.c:32:50: error: 'MP_QSTR_ucollections' undeclared here (not in a function); did you mean 'MP_QSTR_builtins'?
     { MP_ROM_QSTR(MP_QSTR___name__), MP_ROM_QSTR(MP_QSTR_ucollections) },
                                                  ^~~~~~~~~~~~~~~~~~~~
/home/jackson/Projects/esp-mpy-demo/components/mpy/mpy/py/obj.h:96:56: note: in definition of macro 'MP_OBJ_NEW_QSTR'
 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 3) | 2))
                                                        ^~~
/home/jackson/Projects/esp-mpy-demo/components/mpy/mpy/py/modcollections.c:32:38: note: in expansion of macro 'MP_ROM_QSTR'
     { MP_ROM_QSTR(MP_QSTR___name__), MP_ROM_QSTR(MP_QSTR_ucollections) },
                                      ^~~~~~~~~~~
/home/jackson/Projects/esp-mpy-demo/components/mpy/mpy/py/modcollections.c:32:5: warning: missing initializer for field 'value' of 'mp_rom_map_elem_t' {aka 'const struct _mp_rom_map_elem_t'} [-Wmissing-field-initializers]
     { MP_ROM_QSTR(MP_QSTR___name__), MP_ROM_QSTR(MP_QSTR_ucollections) },
     ^
In file included from /home/jackson/Projects/esp-mpy-demo/components/mpy/mpy/py/builtin.h:29,
                 from /home/jackson/Projects/esp-mpy-demo/components/mpy/mpy/py/modcollections.c:27:
/home/jackson/Projects/esp-mpy-demo/components/mpy/mpy/py/obj.h:405:18: note: 'value' declared here
     mp_rom_obj_t value;
                  ^~~~~
In file included from /home/jackson/Projects/esp-mpy-demo/components/mpy/mpy/py/builtin.h:29,
                 from /home/jackson/Projects/esp-mpy-demo/components/mpy/mpy/py/modcollections.c:27:
/home/jackson/Projects/esp-mpy-demo/components/mpy/mpy/py/modcollections.c:36:19: error: 'MP_QSTR_namedtuple' undeclared here (not in a function); did you mean 'MP_QSTR_tuple'?
     { MP_ROM_QSTR(MP_QSTR_namedtuple), MP_ROM_PTR(&mp_namedtuple_obj) },
                   ^~~~~~~~~~~~~~~~~~
/home/jackson/Projects/esp-mpy-demo/components/mpy/mpy/py/obj.h:96:56: note: in definition of macro 'MP_OBJ_NEW_QSTR'
 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 3) | 2))
                                                        ^~~
/home/jackson/Projects/esp-mpy-demo/components/mpy/mpy/py/modcollections.c:36:7: note: in expansion of macro 'MP_ROM_QSTR'
     { MP_ROM_QSTR(MP_QSTR_namedtuple), MP_ROM_PTR(&mp_namedtuple_obj) },
       ^~~~~~~~~~~
[5/104] Building C object esp-idf/mpy/CMakeFiles/__idf_mpy.dir/mpy/py/modarray.c.obj
FAILED: esp-idf/mpy/CMakeFiles/__idf_mpy.dir/mpy/py/modarray.c.obj 
[... cut ...]
 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 3) | 2))
                                                        ^~~
/home/jackson/Projects/esp-mpy-demo/components/mpy/mpy/py/modarray.c:33:7: note: in expansion of macro 'MP_ROM_QSTR'
     { MP_ROM_QSTR(MP_QSTR_array), MP_ROM_PTR(&mp_type_array) },
       ^~~~~~~~~~~
[6/104] Building C object esp-idf/mpy/CMakeFiles/__idf_mpy.dir/mpy/py/modmicropython.c.obj
FAILED: esp-idf/mpy/CMakeFiles/__idf_mpy.dir/mpy/py/modmicropython.c.obj 

```

May I ask have I forgot something? 

Regards,
Jackson

huming2207, 2022-05-03:
From the error it told me, for example, `MP_QSTR___name__` is not declared. However, in the generated `genhdr/qstrdefs.generated.h` header file, there is a a `MP_QSTR___name__`:

```c
// ...
QDEF(MP_QSTR___name__, (const byte*)"\xe2\x08" "__name__")
// ...
```

huming2207, 2022-05-03:
So far the QSTR issue has been fixed, turned out to be I forgot to add the sources into `MICROPY_SOURCE_QSTR`...

However, I've got some more linking errors, no idea for now: 

```
[966/968] Linking CXX executable esp-mpy-demo.elf
FAILED: esp-mpy-demo.elf 
: && /home/hu/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ -mlongcalls  CMakeFiles/esp-mpy-demo.elf.dir/project_elf_src_esp32s3.c.obj -o esp-mpy-demo.elf  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/asio/libasio.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/mdns/libmdns.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/json/libjson.a  esp-idf/mqtt/libmqtt.a  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/usb/libusb.a  esp-idf/ulp/libulp.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/mpy/libmpy.a  esp-idf/main/libmain.a  -Wl,--cref  -Wl,--defsym=IDF_TARGET_ESP32S3=0  -Wl,--Map="/home/hu/Projects/esp-mpy-demo/build/esp-mpy-demo.map"  -fno-rtti  -fno-lto  -Wl,--gc-sections  -Wl,--warn-common  esp-idf/app_trace/libapp_trace.a  -lgcov  esp-idf/app_trace/libapp_trace.a  -lgcov  esp-idf/unity/libunity.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/json/libjson.a  esp-idf/mpy/libmpy.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/mdns/libmdns.a  esp-idf/console/libconsole.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/ulp/libulp.a  esp-idf/mbedtls/mbedtls/library/libmbedtls.a  esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a  esp-idf/mbedtls/mbedtls/library/libmbedx509.a  /home/hu/esp/esp-idf/components/esp_wifi/lib/esp32s3/libcoexist.a  /home/hu/esp/esp-idf/components/esp_wifi/lib/esp32s3/libcore.a  /home/hu/esp/esp-idf/components/esp_wifi/lib/esp32s3/libespnow.a  /home/hu/esp/esp-idf/components/esp_wifi/lib/esp32s3/libmesh.a  /home/hu/esp/esp-idf/components/esp_wifi/lib/esp32s3/libnet80211.a  /home/hu/esp/esp-idf/components/esp_wifi/lib/esp32s3/libpp.a  /home/hu/esp/esp-idf/components/esp_wifi/lib/esp32s3/libsmartconfig.a  /home/hu/esp/esp-idf/components/esp_wifi/lib/esp32s3/libwapi.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs [... cut ...] n `mpy_main::start_repl()':
/home/hu/Projects/esp-mpy-demo/components/mpy/mpy_main.cpp:56: undefined reference to `pyexec_raw_repl()'
/home/hu/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /home/hu/Projects/esp-mpy-demo/components/mpy/mpy_main.cpp:60: undefined reference to `pyexec_friendly_repl()'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

```

luzidchris, 2022-09-24:
Have you made any progress here? I tried to port your project (mpy-demo + mpy idf component) to ESP8266_RTOS_SDK which is like esp-idf. I get the same errors. I first thought this is related to some missing definitions in the mpy config header but this seems to be correct. In my case, there's also an "undefined reference" error related to a FreeRTOS function I had to use. This function is one of the "static allocation" variants, i.e., it depends on configSUPPORT_STATIC_ALLOCATION which is definitely set for the IDF build. It's like as if the definitions "got lost" in the mpy component or some sources for some reason..

fvstory, 2023-05-05:
that is a great idea！！！

closed 2023-07-16 by huming2207

huming2207, 2024-06-05:
> Have you made any progress here? I tried to port your project (mpy-demo + mpy idf component) to ESP8266_RTOS_SDK which is like esp-idf. I get the same errors. I first thought this is related to some [...]

Hi @luzidchris (or any others who saw this),

I've got a solution here: https://github.com/huming2207/soul-injector/tree/next/components/mpy_embed

Basically you need to get the [QSTR](https://docs.micropython.org/en/latest/develop/qstr.html) properly generated. My way of doing it is to let CMake does a `make -f micropython/ports/embed/embed.mk`  to generate the embeddable micropython runtime for my ESP-IDF project with QSTR also generated alongside with the main micropython codebase.