micropython/micropython #8620: ESP32: Crash in mp_binary_op.

RSC-Games, 2022-05-03:
I'm running ESP-IDF 4.2 and using a custom build of MicroPython 1.18.5 on ESP32 Rev 1 without PSRAM.

I wrote a c module to quickly apply a perspective effect to an image (like SNES Mario Kart) and have noticed the following errors just by adding or removing a print statement: (Unfortunately I don't have the backtraces.)
LoadProhibited (qstr_hash) (Only had a couple of times.)
LoadProhibited (mp_binary_op)
InstructionFetchError (Stack Pointer shows mp_binary_op as culprit) 
Only backtrace I have is below:
``` Guru Meditation Error: Core  0 panic'ed (InstructionFetchError). Exception was unhandled.

Core  0 register dump:
PC      : 0x3f4038b4  PS      : 0x00060e30  A0      : 0x80083ece  A1      : 0x3ffc2900  
A2      : 0x00000002  A3      : 0x00000001  A4      : 0x3ffd0cb0  A5      : 0x0000000e  
A6      : 0x0000001e  A7      : 0x00000000  A8      : 0x800dad39  A9      : 0x3f40b7d8  
A10     : 0x00000002  A11     : 0x3ffd0cb0  A12     : 0x00000001  A13     : 0x3f4029c0  
A14     : 0x00000010  A15     : 0x3f4038b4  SAR     : 0x00000019  EXCCAUSE: 0x00000002  
EXCVADDR: 0x3f4038b4  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace:0x3f4038b1:0x3ffc2900 |<-CORRUPTED

ELF file SHA256: 0098e19e7d138f1f ```
InstrFetchProhibited (Stack Pointer shows mp_binary_op as culprit)

From the backtraces, it would appear to me that I would be corrupting data as the function executes properly, but when I do a division to calculate FPS, all of the above crashes have occured. I turned on comprehensive heap corruption detection and stack smashing + stack overflow protection to see if they were the issue. No watchpoint has been triggered. I also ran core dumps and only mp_task is suffering. All other processes are intact.

User C module code is supplied below, as is the Python Module that calls it. If you want the ELF file and the core dump, let me know and I can provide those as well.

Python Code:
``` # Main script of MarioKart SP.
import sys, os, time, render_lib
from framebuf import MONO_HLSB, FrameBuffer
from Console_Iface import iface_select as hw_iface

# This setup should really be done in launch_stub.
hw_iface = hw_iface.get_iface()
hw = hw_iface.hardware
hw.init_hardware()
img_bin = hw_iface.img_bin
screen = hw.display

# Entry point for main.
def main():
	# Select hardware or software rendering.
	render_lib = None
	for i in range(0, 1):
		if hw.buttons[hw.UP_BTN].value() == 0:
			render_lib = __import__("soft_render_lib")
		else:
			render_lib = __import__("render_lib")
		time.sleep(0.1)
	
	# Check if a software renderer is in use.
	if render_lib.render_soft() == True:
		print("WARNING! Slower performance may result from software rendering.")
	else:
		print("WARNING! Instability may result from hardware renderer.")

	# Set useful variables that save execution time.
	HRES = 128
	VRES = 64
	HALF_VRES = VRES // 2
	render_height = 8

	# Set renderer parameters
	render_lib.set_m7_param("hres", 128)
	render_lib.set_m7_param("vres", VRES)
	render_lib.set_m7_param("horizon", 0)
	render_lib.set_m7_param("scaling", 20)
	render_lib.set_m7_param("fov", 200)
	render_lib.set_m7_param("render_height", render_height)
	
[... 337 lines cut ...]
// Table entries are key/value pairs of the attribute name (a string)
// and the MicroPython object reference.
// All identifiers and strings are written as MP_QSTR_xxx and will be
// optimized to word-sized integers by the build system (interned strings).
STATIC const mp_rom_map_elem_t render_lib_module_globals_table[] = {
    { MP_ROM_QSTR(MP_QSTR___name__), MP_ROM_QSTR(MP_QSTR_render_lib) },
    { MP_ROM_QSTR(MP_QSTR_render_plane), MP_ROM_PTR(&render_plane_obj) },
    { MP_ROM_QSTR(MP_QSTR_render_soft), MP_ROM_PTR(&render_soft_obj) },
    { MP_ROM_QSTR(MP_QSTR_spawn_m7_thread), MP_ROM_PTR(&spawn_m7_thread_obj) },
    { MP_ROM_QSTR(MP_QSTR_set_m7_param), MP_ROM_PTR(&set_m7_param_obj) },
};
STATIC MP_DEFINE_CONST_DICT(render_lib_module_globals, render_lib_module_globals_table);

// Define module object.
const mp_obj_module_t render_lib_mod = {
    .base = { &mp_type_module },
    .globals = (mp_obj_dict_t *)&render_lib_module_globals,
};

// Register the module to make it available in Python.
// Note: the "1" in the third argument means this module is always enabled.
// This "1" can be optionally replaced with a macro like MODULE_CPPEXAMPLE_ENABLED
// which can then be used to conditionally enable this module.
MP_REGISTER_MODULE(MP_QSTR_render_lib, render_lib_mod, 1);
```

dpgeorge, 2022-05-03:
> wndw_ptr = MP_OBJ_TO_PTR(out_img);

This won't work, you need to use `mp_get_buffer_raise(out_img, &bufinfo, MP_BUFFER_RW)`.

dpgeorge, 2022-05-03:
Otherwise, this doesn't look like a problem with MicroPython itself, but rather with the C module you have written.  The best place to ask questions about usage is at https://forum.micropython.org

closed 2022-05-03 by dpgeorge

RSC-Games, 2022-05-04:
Thanks for your time! I got around to fixing the bug today, and the code doesn't crash anymore! However, I'm still experiencing issues related to my lack of knowledge of how FrameBuffer handles pixel writes to MONO_HLSB images. Here's the link: [](https://forum.micropython.org/viewtopic.php?f=18&t=12372)