micropython/micropython #8621: improving timer accuracy on macos

dlech, 2022-05-04:
In #8617 we tried using `nice` to try to get more accurate timers to fix failing tests. This didn't work. However there does seem to be a way to give threads realtime priority to get more accurate timers.

>  I did some digging at it appears that `nice` does nothing on macos. [1](https://stackoverflow.com/questions/44811425/nice-command-not-working-on-macos) [2](https://unix.stackexchange.com/questions/384385/macos-nice-renice-not-working-no-actual-effect-on-cputime) [3](https://unix.stackexchange.com/questions/442470/macos-10-13-4-renice-seems-to-have-no-effect) I also confirmed this with my own measurements (a simple C program with `usleep`).
>
> It appears that to get accurate timing on macos we need to make threads realtime threads: https://developer.apple.com/library/archive/technotes/tn2169/_index.html
>
> I tried the example code from the tech note and it does make a difference.

_Originally posted by @dlech in https://github.com/micropython/micropython/issues/8617#issuecomment-1116863210_

In the tests I did, the current implementation (`usleep(500)` with default thread priority) the actual time measured for 10000 trials varied between 500 and 5000 microseconds. With the example code from the tech note (setting the thread to realtime priority and using `mach_wait_until` instead of `usleep`), the time was < 600 microseconds 99.9% of the time with the longest time being just shy of 1000 microseconds. I also got the same results using reatime priority with `usleep`.

I'm not sure that we would always want MicroPython to run with realtime priority but perhaps it makes sense to add a `-x` option to enable it from the command line for uses who know what they are doing and to help with failing tests.

dpgeorge, 2022-05-04:
> I'm not sure that we would always want MicroPython to run with realtime priority but perhaps it makes sense to add a `-x` option to enable it from the command line for uses who know what they are doing and to help with failing tests.

Yes that could be a good thing to do, an option like `micropython -X realtime`.

dpgeorge, 2022-05-23:
Done in c012318d74fed232b1c872a1a0cacf91ebcb1b6d