micropython/micropython #8627: Cannot disable CONFIG_ESP_NETIF_TCPIP_ADAPTER_COMPATIBLE_LAYER in sdkconfig

geza-pycom, 2022-05-04:
Hello,

When MICROPY_PY_NETWORK_WLAN is set to 0, it means WLAN is not used, however it is still required to keep CONFIG_ESP_NETIF_TCPIP_ADAPTER_COMPATIBLE_LAYER set to TRUE in sdkconfig otherwise compilation errors happen in modnetwork.c.

When MICROPY_PY_NETWORK_WLAN is off, no WLAN specific items shall be included in the build, or at least the esp32 port's WLAN implementation shall be updated to use the new esp_event APIs from esp-idf introduced in esp-idf 4.0, tcpip_adapter is marked as deprecated.

jonnor, 2024-09-08:
In latest git master, I no longer see any references to CONFIG_ESP_NETIF_TCPIP_ADAPTER_COMPATIBLE_LAYER

I believe that the code has been refactored to use the new esp_event api, as we now use esp-idf 5.x+
So perhaps this issue has been fixed?

dpgeorge, 2024-09-08:
The IDF has been updated since this issue was opened, so I think we can close this issue.

It can be reopened if it's still a problem.

closed 2024-09-08 by dpgeorge