micropython/micropython #8628: ESP32 OSError: [Errno 12] ENOMEM urequests

chriscmorgan, 2022-05-04:
I am getting the ENOMEM error when trying to use urequests to get data from our https api server. I am using the esp32-c3-wroom-2 (4MB RAM).

` 
import bluetooth
ble = bluetooth.BLE()
ble.active(True)
config = station.ifconfig()
print (config[0])

request_url = 'https://<api server>' 
gc.collect()
print("Memory Free-8")
print(gc.mem_free())
print("before api request")

res = urequests.post(request_url, headers = {'content-type': 'application/json'}, data = post_data)
print(res)
print("after api request")`

Error:
`Try connection
192.168.137.238
Memory Free-8
91408
before api request
Traceback (most recent call last):
  File "<stdin>", line 77, in <module>
  File "/mnt/c/Users/chris.morgan/Documents/code/ilume-hub-firmware/lib/urequests.py", line 60, in request
OSError: [Errno 12] ENOMEM`

This error occurs only occurs on esp32c3-20220117-v1.18.bin but on esp32c3-20210902-v1.17.bin is works without error.
If I disable BLE it also works on 1.18 but I need it enabled all the time for my application.
I want to use 1.18 as 1.17 only enabled half the memory that 1.18 does and my application requires  more memory.

Any help is really appreciated.

chriscmorgan, 2022-05-09:
OK this actually seems to go with the espidf version and not the micropython version.
espift 4.3.2 or 4.4 creates this issue. There is more memory available for my python code to run but you can't have urequests and ble run at the same time. espidt4.3 works fine with both running but half the memory,
I can see 1.17 is compiled with 4.3 at: https://micropython.org/download/esp32c3/
while 1.18 is compiled with 4.4.
Any suggestions on how I can get the extra micropython memory while still being able to use urequests and ble?

chrisovergaauw, 2022-06-03:
I think your last question is better placed at the micropython forum to be honest.
I've ran into similar issues and dabbled with the mp_task_heap_size below to get my setup working.
https://github.com/micropython/micropython/blob/master/ports/esp32/main.c#L142

It's not really a fix and resulted in a "magic variable", so use it as a last resort.
I'm not relying on anything specific in the v4.4 so I downgraded to 4.2.3 (notice that's not 4.3.2 which you mentioned) and had a lot more success with running ble and network requests concurrently.

puppet13th, 2022-07-16:
i think i have same issue regarding ble on esp32-c3 port.
my code uses ble + asyncio
there is no issue when running the code using esp32 port(firmware 1.18 and 1.19.1)
there is weird issue when running the code using esp32-c3 port(firmware 1.18 and 1.19.1)
now testing esp32-c3 running on firmware 1.17 -> looks good.
please check my forum [post at](https://forum.micropython.org/viewtopic.php?f=18&t=12696)

ciprianmisaila, 2022-12-20:
edit urequest.py. It does not close the connection at the end of request function. Too many sockets opened.

hezi-biancheng, 2024-06-16:
> 当我尝试使用 urequests 从我们的 https api 服务器获取数据时，我收到 ENOMEM 错误。仅当 BLE 同时处于活动状态时，才会发生错误。这在 1.17 中没有发生，但现在使用 1.18 发生了。我需要 1.18 提供的额外内存，所以有点束缚。我正在使用 esp32-c3-wroom-2 （4MB RAM）。
> 
> ```
> import bluetooth
> ble = bluetooth.BLE()
> ble.active(True)
> config = station.ifconfig()
> print (config[0])
> 
> request_url = 'https://<api server>' 
> gc.collect()
> print("Memory Free-8")
> print(gc.mem_free())
> print("before api request")
> 
> res = urequests.post(request_url, headers = {'content-type': 'application/json'}, data = post_data)
> print(res)
> print("after api request")
> ```
> 
> 错误：
> 
> ```
> 192.168.137.238
> Memory Free-8
> 91408
> before api request
> Traceback (most recent call last):
>   File "<stdin>", line 77, in <module>
>   File "/mnt/c/Users/chris.morgan/Documents/code/ilume-hub-firmware/lib/urequests.py", line 60, in request
> OSError: [Errno 12] ENOMEM
> ```
> 
> 此错误仅在esp32c3-20220117-v1.18.bin上发生，但在esp32c3-20210902-v1.17.bin上正常工作没有错误。如果我禁用 BLE，它也可以在 1.18 上运行，但我需要一直为我的应用程序启用它。我想使用 1.18，因为 1.17 只启用了 1.18 的一半内存，我的应用程序需要更多的内存。
> 
> 任何帮助都非常感谢。
我也遇到了这个问题，但是我并没有打开蓝牙，在第一次连接时没有问题，第二次就出问题了，我现在特别苦恼，下面是我的导入列表：import ufont
import urequests
import utime
import random
import time
from machine import Timer
from st7735 import ST7735
from machine import SPI, Pin

qfbt, 2024-10-11:
I met the similar error today, I finally found that it was the insuficiency of power cause it ：
I used the expansion board,but I only gave power to esp32 for convenience.
That behavior may weaken the esp32's stability, so when I put sendto() in a loop, it corrupted in few rounds.
so I recommend to check the power supply

qfbt, 2024-10-11:
> I met the similar error today, I finally found that it was the insufficiency of power cause it ：
> I used the expansion board,but I only gave power to esp32 for convenience.
> That behavior may weaken the esp32's stability, so when I put sendto() in a loop, it corrupted in few rounds.
> so I recommend to check the power supply

qfbt, 2024-10-11:
> 当我尝试使用 urequests 从我们的 https api 服务器获取数据时，我收到 ENOMEM 错误。仅当 BLE 同时处于活动状态时，才会发生错误。这在 1.17 中没有发生，但现在使用 1.18 发生了。我需要 1.18 提供的额外内存，所以有点束缚。我正在使用 esp32-c3-wroom-2 （4MB RAM）。
> ```
> import bluetooth
> ble = [...]

检查一下供电稳定性