micropython/micropython #8629: Supporting asyncio with modusocket/BSD-like sockets.

iabdalkader, 2022-05-05:
Hi, I'm working on a new BSD-like sockets ABI for the Nina-W10 module and an updated driver/module, and I want to make sure it passes all (or most) of the multi_net tests, especially asyncio, before committing it. However, it seems there is a specific (maybe non-standard?) behavior expected from sockets functions called from `asyncio` or from higher layers. To be more specific, this test `multi_net/uasyncio_tcp_readinto.py` for example, fails with the following DIFF output:

```python
 --- instance1 ---
 2
 bytearray(b'ab')
-1
-array('b', [99])
-TypeError
-TypeError
+
+Traceback (most recent call last):
+  File "<stdin>", line 128, in <module>
+  File "<stdin>", line 80, in instance1
+  File "uasyncio/core.py", line 1, in run
+  File "uasyncio/core.py", line 1, in run_until_complete
+  File "uasyncio/core.py", line 1, in run_until_complete
+  File "<stdin>", line 58, in tcp_client
+  File "uasyncio/stream.py", line 1, in readinto
+OSError: [Errno 107] ENOTCONN
```
The output makes sense to me because by the time the second `n = await reader.readinto(a)` is called from the client side, the socket is already closed by the server, and the client gets `ENOTCONN` from `recv()` as expected, but for some reason `asyncio` or doesn't like/expect that, and I'm not sure what `recv()` should do here ?

To get around this issue and keep testing, I add a `await asyncio.sleep(1.0)` just before `writer.close()` in `handle_connection`, and I see some valid output, but then I get this DIFF:

```python
### DIFF ###
--- /tmp/tmp6cq8l7qi	2022-05-05 12:32:05.560643907 +0200
+++ /tmp/tmp08ybaqok	2022-05-05 12:32:05.560643907 +0200
@@ -7,5 +7,4 @@
 bytearray(b'ab')
 1
 array('b', [99])
-TypeError
-TypeError
+TIMEOUT
```

It seems that `n = await reader.readinto(5)` keeps calling `socket_ioctl()` to poll the closed socket which returns 0, and never gets to the point where it raises a `TypeError` and `run-multitests.py` timesout. Again, I'm not sure what is `socket_ioctl()` expected to do/return in this case.

dpgeorge, 2022-05-05:
> I'm working on a new BSD-like sockets ABI for the Nina-W10 module

Did you consider a raw Ethernet frame based ABI?  So the TCP/IP stack could run on the host?  Then you could use `modlwip.c` and get all the correct semantics for free.

> The output makes sense to me because by the time the second `n = await reader.readinto(a)` is called from the client side, the socket is already closed by the server, and the client gets `ENOTCONN` from `recv()` as expected

The server writes 3 bytes (2 bytes then 1 byte in separate calls) then closes the socket.  That means the client should see 3 bytes pending on its socket.  Regardless of whether the socket is closed, there should still be 3 bytes available for the application to read.  The application should not need to be fast enough to read these 3 bytes before the client closes the connection.  (Internally the TCP/IP stack will have read and ack'd those 3 bytes, buffered them, then ack'd the closure of the connection.  And had those 3 bytes ready for the application to read out.)

iabdalkader, 2022-05-05:
>> Did you consider a raw Ethernet frame based ABI

Yes, it would be my first choice too, but after a discussion with Arduino, they prefer to have a sockets-like ABI added along side the older commands, so that it can be used with resource constrained "hosts" that can Not run lwip. Other than that, technically speaking I'm not sure how to implement it (or if it's even possible) because the firmware already runs lwip, and it registers all sorts of callbacks etc. admittedly thought I didn't look hard enough. One thought occured to me, would it be possible to open a RAW socket and use it for that ?

>>  The application should not need to be fast enough to read these 3 bytes before the client closes the connection

There seems to be some timing issue here, because if I add the sleep I get the 3 bytes like in the second DIFF output. Other than that, anything special `socket_ioctl` should do ?

dpgeorge, 2022-05-05:
> One thought occured to me, would it be possible to open a RAW socket and use it for that ?

Yes that could be a possibility, but would need investigation to see if the ETH packets really do go in/out without modification.

> Other than that, anything special `socket_ioctl` should do ?

You can look at the logic in `modlwip.c:lwip_socket_ioctl`:
- if there are bytes available, the socket is readable (even if closed)
- if the socket is open and has free space then it's writable
- if the socket is closed then it's always readable and writable

dpgeorge, 2022-05-06:
> This could be an issue though because I don't have any access to the lwip socket state from the BSD sockets API.

Can't you just use select on the lwip socket?  The BSD sockets API should have a way to poll the socket, and you can just make a thin wrapper between that and the socket_ioctl on the host side.

iabdalkader, 2022-05-06:
> Can't you just use select on the lwip socket

I can, if it has the same semantics as `modlwip.c:lwip_socket_ioctl()` then it should work. I was trying to provide the standard ioctl with the options available in lwip.

iabdalkader, 2022-05-06:
Getting there :smile: 

```bash
multi_net/uasyncio_tcp_close_write.py.exp on : pass
multi_net/uasyncio_tcp_readexactly.py on ttyACM0|micropython: pass
multi_net/uasyncio_tcp_readexactly.py.exp on : pass
multi_net/uasyncio_tcp_readinto.py on ttyACM0|micropython: pass
multi_net/uasyncio_tcp_readinto.py.exp on : pass
multi_net/uasyncio_tcp_server_client.py on ttyACM0|micropython: pass
multi_net/uasyncio_tcp_server_client.py.exp on : pass
10 tests performed
9 tests passed
1 tests failed: multi_net/uasyncio_tcp_close_write.py
```

iabdalkader, 2022-05-11:
Also, FYI the following test fails on `PYBD_SF2`:

```bash
 ./run-multitests.py -i pyb:a0 multi_net/tcp_client_rst.py 
multi_net/tcp_client_rst.py on ttyACM0|micropython: FAIL
### TEST ###
--- instance0 ---
[16]
104
[32]

Traceback (most recent call last):
  File "<stdin>", line 104, in <module>
  File "<stdin>", line 39, in instance0
OSError: [Errno 9] EBADF

--- instance1 ---

### TRUTH ###
--- instance0 ---
[25]
b'GET / HTTP'
[25]
b'/1.0\r\n\r\n'
[25]
104
[17]
b''
[17]
--- instance1 ---

### DIFF ###
--- /tmp/tmp5sn75yal	2022-05-11 09:06:55.802083335 +0200
+++ /tmp/tmpmvx4lxc2	2022-05-11 09:06:55.802083335 +0200
@@ -1,12 +1,12 @@
 --- instance0 ---
-[25]
-b'GET / HTTP'
-[25]
-b'/1.0\r\n\r\n'
-[25]
+[16]
 104
-[17]
-b''
-[17]
+[32]
+
+Traceback (most recent call last):
+  File "<stdin>", line 104, in <module>
+  File "<stdin>", line 39, in instance0
+OSError: [Errno 9] EBADF
+
 --- instance1 ---
 
1 tests performed
0 tests passed
1 tests failed: multi_net/tcp_client_rst.py
```

iabdalkader, 2022-05-26:
Fixed in #8654

closed 2022-05-26 by iabdalkader