micropython/micropython #8634: BUG on the ESP32-S3 Hardware SPI

beyonlo, 2022-05-05:
Hi all.

I'm using the ESP32-S3-WROOM-1 (N8R2 - 8MB Flash and 2 MB RAM), specifically this kit: [ESP32-S3-DEVKITC-1-N8R2](https://www.digikey.com/en/products/detail/espressif-systems/ESP32-S3-DEVKITC-1-N8R2/15199627)

The hardware SPI are working just on the **number 2**. 

As I know, the ESP32-S3, like as the ESP32 has two free Hardware SPI to use (SPI 1 and SPI 2). It has 4 in total, but two SPI (SPI 3 and SPI 4) are used by the RAM/FLASH. So, should works Hardware SPI number 1 and number 2, right?

**Follow the tests:**

```
>>> import os
>>> os.uname()
(sysname='esp32', nodename='esp32', release='1.18.0', version='v1.18 on 2022-05-05', machine='ESP32S3 module (spiram) with ESP32S3')
>>> 
>>> from machine import SPI, Pin
>>> 
>>> spi  = SPI(1, baudrate=5_000_000, phase=1, sck=Pin(15), mosi=Pin(16), miso=Pin(17))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: invalid configuration
>>> 
>>> spi  = SPI(2, baudrate=5_000_000, phase=1, sck=Pin(15), mosi=Pin(16), miso=Pin(17))
>>> 
>>> spi  = SPI(3, baudrate=5_000_000, phase=1, sck=Pin(15), mosi=Pin(16), miso=Pin(17))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: SPI(3) doesn't exist
>>> 
>>> spi  = SPI(4, baudrate=5_000_000, phase=1, sck=Pin(15), mosi=Pin(16), miso=Pin(17))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: SPI(4) doesn't exist
>>> 
```

Follow below the same test with the ESP32:

```
>>> import os
>>> os.uname()
(sysname='esp32', nodename='esp32', release='1.18.0', version='v1.18 on 2022-01-17', machine='ESP32 module with ESP32')
>>> 
>>> from machine import SPI, Pin
>>> spi  = SPI(1, baudrate=5_000_000, phase=1, sck=Pin(15), mosi=Pin(16), miso=Pin(17))
>>> 
>>> spi  = SPI(2, baudrate=5_000_000, phase=1, sck=Pin(15), mosi=Pin(16), miso=Pin(17))
>>> 
>>> spi  = SPI(3, baudrate=5_000_000, phase=1, sck=Pin(15), mosi=Pin(16), miso=Pin(17))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: SPI(3) doesn't exist
>>> spi  = SPI(4, baudrate=5_000_000, phase=1, sck=Pin(15), mosi=Pin(16), miso=Pin(17))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: SPI(4) doesn't exist
>>> 
```

I miss some configuration or is really a BUG?

Thank you.

robert-hh, 2022-05-06:
What happens with SPI(1), if you use different pins. I did not look into the schematics of that device, but at the 'old' ESP3 devices pins 16 and 17 were used for internal flash, and to  protect people from crashed, use of that pins was rejected. So maybe this code is still used, but not needed any more. Same for using SPI 2 and 3.

beyonlo, 2022-05-06:
Hello @robert-hh 

> What happens with SPI(1), if you use different pins. I did not look into the schematics of that device, but at the 'old' ESP3 devices pins 16 and 17 were used for internal flash, and to protect [...]

Happens the same error. I tried 3 different Pins and all works on SPI(2) but not works on SPI(1):

```
>>> import os
>>> os.uname()
(sysname='esp32', nodename='esp32', release='1.18.0', version='v1.18 on 2022-05-05', machine='ESP32S3 module (spiram) with ESP32S3')
>>> from machine import SPI, Pin
>>> spi  = SPI(1, baudrate=5_000_000, phase=1, sck=Pin(9), mosi=Pin(10), miso=Pin(11))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: invalid configuration
>>> spi  = SPI(2, baudrate=5_000_000, phase=1, sck=Pin(9), mosi=Pin(10), miso=Pin(11))
>>> 
>>> spi  = SPI(1, baudrate=5_000_000, phase=1, sck=Pin(12), mosi=Pin(13), miso=Pin(14))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: invalid configuration
>>> spi  = SPI(2, baudrate=5_000_000, phase=1, sck=Pin(12), mosi=Pin(13), miso=Pin(14))
>>> 
>>> spi  = SPI(1, baudrate=5_000_000, phase=1, sck=Pin(5), mosi=Pin(6), miso=Pin(7))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: invalid configuration
>>> spi  = SPI(2, baudrate=5_000_000, phase=1, sck=Pin(5), mosi=Pin(6), miso=Pin(7))
>>> 

```

stephanelsmith, 2022-05-18:
I define both the SPI1 and SPI2 pins in my mpconfigboard.h file.  For example, in the repo, check out SPI1 definitions in UM_TINYS3/mpconfigboard.h
```
#define MICROPY_HW_SPI1_MOSI                (35)
#define MICROPY_HW_SPI1_MISO                (37)
#define MICROPY_HW_SPI1_SCK                 (36)
```

stephanelsmith, 2022-05-18:
From the spec sheet:
ESP32-S3 features four SPI interfaces (SPI0, SPI1, SPI2 and SPI3). SPI0 and SPI1 can be configured to operate
in SPI memory mode; SPI2 and SPI3 can be configured to operate in general-purpose SPI mode.

I use R8, which means SPI1 is in memory mode, however, I define both the SPI1 and SPI2 pins in my mpconfigboard.h file. For example, in the repo, check out SPI1 definitions in UM_TINYS3/mpconfigboard.h
#define MICROPY_HW_SPI1_MOSI                (35)
#define MICROPY_HW_SPI1_MISO                (37)
#define MICROPY_HW_SPI1_SCK                 (36)

beyonlo, 2022-05-18:
@stephanelsmith Hello, thanks for your reply!

> From the spec sheet: ESP32-S3 features four SPI interfaces (SPI0, SPI1, SPI2 and SPI3). SPI0 and SPI1 can be configured to operate in SPI memory mode; SPI2 and SPI3 can be configured to operate in general-purpose SPI mode.
> 
> This is a little different than what you mention in your initial comment.

Yes, what I mention is a bit different because is really different what I'm talking about than you are talking about. You are talking about Microcontroller definitions, and I'm talking about MicroPython definitions.

The MicroPython definitions say that ESP32 has two free SPI to use https://docs.micropython.org/en/latest/esp32/quickref.html#hardware-spi-bus That two SPI are mapped (by MicroPython) as SPI(1) and SPI(2). I think that MicroPython does that remap because it is a better way than to follow the datasheet map, so the user can to use the two free SPI on MicroPython source code as SPI(1) and SPI(2), Will be a weird to use SPI(3) and SPI(4) on Python applications and do not exists the SPI(1) and SPI(2).

Look above in my test that I tested the SPI(3) and SPI(4), on the ESP32 and ESP32-S3 too, and always the answer is `ValueError: SPI(4) doesn't exist` confirming that the two free SPI to use is SPI(1) and SPI(2).  The SPI(2) works fine, but when tried the SPI(1) the error is `OSError: invalid configuration`. And, in my tests above I tried the SPI(1) in many other pins combinations and always the same error.

> 
> I use R8, which means SPI1 is in memory mode (octal RAM configuration), so I cannot test SPI1, however, check out SPI1 definitions in UM_TINYS3/mpconfigboard.h #define MICROPY_HW_SPI1_MOSI (35) #define MICROPY_HW_SPI1_MISO (37) #define MICROPY_HW_SPI1_SCK (36)
> 
> FSPID, FSPICLK, FSPIQ are defined on pin 35,36,37 and 11,12,13. I know there's a cross bar in there and you should be able to make anything go to anything, but haven't looked into that deeply. Would be curious on your findings.

As described in ESP32-S3 datasheet, the ESP32 support to use SPI in any pins, but hardware SPI channels are limited to 40MHz when used on pins other than the described on datasheet. So, for this test, just to check if the two free SPI are working on the ESP32-S3, any pins can be used.

beyonlo, 2022-08-15:
Did anyone success to work `SPI(2)` on the ESP32-S3? Just `SPI(1)` works!

victorallume, 2022-10-04:
I'm hitting the same problem on the S3 on micropython 1.19.1

dpgeorge, 2022-10-04:
I can confirm the bug, using a UM_TINYS3 board and GENERIC_S3 firmware.  The problem is that S3 chips need to use auto DMA configuration.  Should be fixed by #9498.

dpgeorge, 2022-10-05:
Fixed by 95c614e2b60b5c1c83c85aefe88a2aadbf43b9ed

closed 2022-10-05 by dpgeorge

Aleks-Falcon, 2022-12-29:
I have assembled the firmware with a new file: ports/esp32/machine_hw_spi.c

As a result , I received:
>>> spi1  = SPI(1, baudrate=5_000_000, phase=1, sck=Pin(5), mosi=Pin(6), miso=Pin(7))	ok!	
>>> spi2  = SPI(2, baudrate=5_000_000, phase=1, sck=Pin(12), mosi=Pin(13), miso=Pin(14))	ok!

but if you write like this:
>>> spi1  = SPI(1)	ok!
>>> spi2  = SPI(2)	error!!! Process ended with exit code 1.

What is my mistake?

rvinuela, 2025-04-24:
I have the same issue with SPI 1, sometimes work after soft rebooting the device. I checked the source code and SPI1 and SPI2 are mapped to SPI_2 and SPI_3, so it should not be a problem of selecting SPI0 and SPI1, or at least is how its written in the comments in machine_hw_spi.c:
```
// SPI mappings by device, naming used by IDF old/new
// upython   | ESP32     | ESP32S2   | ESP32S3 | ESP32C3 | ESP32C6
// ----------+-----------+-----------+---------+---------+---------
// SPI(id=1) | HSPI/SPI2 | FSPI/SPI2 | SPI2    | SPI2    | SPI2
// SPI(id=2) | VSPI/SPI3 | HSPI/SPI3 | SPI3    | err     | err
```
But if I use SPI id=2 (SPI3) there is not problem.

I also added / tested a custom module in C (ESP IDF) to the micropython compilation using directly SPI id=1 (SPI2) and it works without problems, so it might be software related.