micropython/micropython #8635: Possible bug with WiFi on ESP32-S3

beyonlo, 2022-05-05:
Hi all.

I'm using the ESP32-S3-WROOM-1 (N8R2 - 8MB Flash and 2 MB RAM), specifically this kit: [ESP32-S3-DEVKITC-1-N8R2](https://www.digikey.com/en/products/detail/espressif-systems/ESP32-S3-DEVKITC-1-N8R2/15199627)

I tested the `STA` and `AP` mode, in the both mode ESP32-S3 just congeals/stop to to works when the WLAN is activated -> `wlan.active(True)`

**Testing WiFi as STA mode:** 

```
>>> import os
>>> os.uname()
(sysname='esp32', nodename='esp32', release='1.18.0', version='v1.18 on 2022-05-05', machine='ESP32S3 module (spiram) with ESP32S3')
>>> 
>>> import network
>>> wlan = network.WLAN(network.STA_IF) # create station interface
>>> wlan.active(True)       # activate the interface
Terminating...
Thanks for using picocom

```

**Testing WiFi as AP mode:** 

```
>>> import os
>>> os.uname()
(sysname='esp32', nodename='esp32', release='1.18.0', version='v1.18 on 2022-05-05', machine='ESP32S3 module (spiram) with ESP32S3')
>>> 
>>> import network
>>> ap = network.WLAN(network.AP_IF) # create access-point interface
>>> ap.config(essid='ESP-AP') # set the ESSID of the access point
>>> ap.config(max_clients=10) # set how many clients can connect to the network
>>> ap.active(True)         # activate the interface
Terminating...
Thanks for using picocom

```

 Is it really a bug?

Thank you!

Lobo-T, 2022-05-10:
I can confirm that Wifi works on the ESP32-S3.

If it resets just when you activate wifi it seems like you may have a power supply problem.  It draws quite a lot of current just as the Wifi or Bluetooth starts.
How are you supplying power?  If it's through the USB port.  Do you have a very long cable?  Do you go through a hub?

beyonlo, 2022-05-10:
> I can confirm that Wifi works on the ESP32-S3.
>
> If it resets just when you activate wifi it seems like you may have a power supply problem. It draws quite a lot of current just as the Wifi or [...]

@Lobo-T Hello.

The device do not resets, it just stop (congeals) when I ativate it: `wlan.active(True)`

My cable is exactly the same used on the ESP32 (that works fine) - I tried others cables too. It is the standard cable, and it is connected directly to notebook UBS, that delivery until 1.5A per port.

Could you please, paste here your firmware version, your activating process and what ESP32-S3 hardware version are you using?

Thank you very much.

beyonlo, 2022-05-12:
Hi @Lobo-T 

I found the problem. That error happen just when USB port is used on the kit, and I always use USB. If I use UART port on the kit, works fine. Could you please to confirm that that WiFi problem happen using the USB port?

Thank you.

Lobo-T, 2022-05-12:
The above interaction was through REPL on the USB port.

beyonlo, 2022-05-12:
@Lobo-T 

Very strange! You are using the kit ESP32-S3-DEVKITC-1-N8R8 and I using the theESP32-S3-DEVKITC-1-N8R2. The difference is just the RAM R2 vs R8. Maybe a problem just for R2?

I tried to connect using a USB hub too and the same results, many cables tested too. That error happen just using the USB. I'm using Ubuntu 20.04.

Any suggestion?

Lobo-T, 2022-05-12:
Are you using the latest version?  Building from source or using the nightly build?  Or are you using the now quite old "stable" release?

beyonlo, 2022-05-12:
> Are you using the latest version? Building from source or using the nightly build? Or are you using the now quite old "stable" release?

@Lobo-T I'm was using the last stable 1.18. But I tried the nightly build too:

```
>>> import os
>>> os.uname()
(sysname='esp32', nodename='esp32', release='1.18.0', version='v1.18 on 2022-05-06', machine='ESP32S3 module with ESP32S3')
>>>
```

beyonlo, 2022-05-12:
> Are you using the latest version? Building from source or using the nightly build? Or are you using the now quite old "stable" release?

@Lobo-T I'm not building from source. I using ready firmware from here: https://micropython.org/download/GENERIC_S3_SPIRAM/

Lobo-T, 2022-05-13:
I have two ESP32-S3-DEVKITC-1-N8R8 and one ESP32-S3-DEVKITC-1-N8 (no SPIRAM).
They all work with Wifi, either when connected to REPL through USB or through UART.
This is with Micropython firmware I have compiled myself from latest MASTER.
I think I had to make one change to get the 8MB SPIRAM to work, as it's connected through octal SPI, not quad.  But as far as I remember that made it not boot at all, and your version with 4MB should be quad.

If you want to you can try my firmware file for the version without SPIRAM, it should work even though your board actually has SPIRAM: https://drive.google.com/file/d/1kfFMO7f0gfNMf0w2LCkJYp8RwxMiDCgB/view?usp=sharing

Otherwise I think we must conclude that there is something wrong with your particular devkit.

beyonlo, 2022-05-13:
Hello @Lobo-T 

> I have two ESP32-S3-DEVKITC-1-N8R8 and one ESP32-S3-DEVKITC-1-N8 (no SPIRAM). They all work with Wifi, either when connected to REPL through USB or through UART. This is with Micropython firmware I [...]

I flashed my ESP32-S2 with your firmware and WiFi works fine over USB :). Thank you very much.

So, problem is the default MP firmware that as some problem with USB + WiFi. Maybe this confirm that is a BUG on latest ready MP binary version (1.18)?

Thank you.

Lobo-T, 2022-05-17:
I don't think I can help.
The only change I have done is changing [this line](https://github.com/micropython/micropython/blob/1fb01bd6c5dc350f3c617ca8edae8dea9e5516ae/ports/esp32/boards/sdkconfig.spiram_sx#L4) :
From: CONFIG_SPIRAM_MODE_QUAD=y
To: CONFIG_SPIRAM_MODE_OCT=y

stephanelsmith, 2022-05-18:
S3 seems to be a little bit more sensitive than S2 and previous versions, but I use S3-R8 and R2 wifi/ble without any issues, both UART and USB on my own firmware built from master.  I think there's something else going on that may be more specific to your setup.

beyonlo, 2022-05-18:
@stephanelsmith Hello

> S3 seems to be a little bit more sensitive than S2 and previous versions

What do you mean exactly with `sensitive`?

, but I use S3-R8 and R2 wifi/ble without any issues, both UART and USB on my own firmware built from master on my own hardware. I think there's something else going on that may be more specific to your setup. I've also tried a bunch of different S3 series devices from Unexpected Maker and have had success with wifi in a number of different configurations.
> 
> Again, admittedly s3 is clearly more sensitive when it comes to wifi in terms of connection, but I have not yet heard or seen errors when activating sta. What I've seen a bunch is not able to connect to an ap.

That problem happen just on the standard MicroPython firmware (from download pages), not when it is build by yourself - look messages above.

dpgeorge, 2022-06-18:
I can confirm that ESP32-S3 firmware from the download page has issues with WLAN.actiev(1).  I tested on a TinyS3 with v1.19.1 firmware.  Building locally, it works.

I think the problem is that the download build server is using an older IDF v4.4 version.  I have now updated it to use IDF v4.4.1.  It's in the process of rebuilding the firmware now, using this newer IDF.

dpgeorge, 2022-06-20:
I have tested the latest v1.19.1 build from the download page (GENERIC_S3 on a TinyS3) and WLAN works.

closed 2022-06-20 by dpgeorge

beyonlo, 2022-06-20:
I can confirm that GENERIC_S3 version v1.19.1 build from the download page is now working with WiFi on USB port.

Thank you!

MATTYGILO, 2022-09-21:
I am having problems with SPIRAM esp32 s3. It works on GENERIC for me but not spiram version.

jrbecart, 2024-02-13:
In my case, I had issues with two ESP32-S3 I was able to see my 2 networks, but I couldn't connect to them.
Changing the MAC address resolved my issues.

demitrix, 2024-05-24:
> ~In my case, I had issues with two ESP32-S3 I was able to see my 2 networks, but I couldn't connect to them. Changing the MAC address resolved my issues.~
> 
> Forget this previous comment. I just found the crazy reason: Wifi connection problem is only when board is connected to breadboards. Problem does not occur with my other ESP32 (without ceramic antenna) I tested multiple times and I am now 100% sure.

Oddly enough this is my same issue. Off breadboard its fine. Even with the antenna as close to the edge as possible, no connection.

peterhinch, 2024-05-25:
Some boards with ESP32-S3 have weird RF issues. You might try using a low WiFi channel number and/or reducing TX power. See [this thread](https://github.com/orgs/micropython/discussions/12017).