micropython/micropython #8641: ESP32-C3 I2S

dave1167, 2022-05-06:
Hi, 
Is there any technical reason why I2S is missing from the ESP32-C3 build?  Is there a likelihood it will be included in the future?
Many thanks

andypiper, 2022-05-09:
It looks like I2S support for the C3 was incomplete in ESP-IDF itself at the time of the initial commits to MicroPython, but it [arrived](https://github.com/espressif/esp-idf/commit/8131d6f46d690e9cb60cc5cd457863cc5479351f) shortly afterwards. In principle it could just be a case of toggling it on in the config file for the base ESP32-C3 build and testing it. I don’t have anything that I know of in my kit box that is specific for I2S, otherwise I’d be happy to help here!

ricksorensen, 2023-06-22:
I have been fiddling with I2S using the Seeed XIAO ESP32C3 (GPIO 6,7,8).  
1.  To get a good compile with (e.g.) 1.20.0 code modified `machine_i2s.c` to:
     ` #define DR_REG_I2S_BASE                        0x6002D000` 
      before `i2s.h` include. and modifed the conditional around Mike Teachman's bug fix mod:
      `#if CONFIG_IDF_TARGET_ESP32S3 || CONFIG_IDF_TARGET_ESP32C3`
     The code then compiles (GENERIC_C3_USB with I2S enabled).  And runs (note caveats below)
2.  Using idf.py version 4.4.4/4.4.5 I then removed all the bug fix mods as it appears they had been addressed in the IDF SDK.
      The code compiles and runs.

Caveats:
I tried the i2s examples `easy_wav_player` and `play_tone`  by copying those files to the MCU- including wav files.  I also tried with both binaries with the same results.  I am using a generic PCM5102 board to drive audio.

`mpremote exec "import play_tone"` and `mpremote exec "import easy_wav_player"` both worked.  However if I go into the REPL,
`>>> import easy_wav_player` starts with just noise out when playing stereo wav, and then mono wav works.  Rerunning (without reset) and then the stereo wav plays too.
If I restart REPL, and 
`>>>import play_tone' first then `>>>import easy_wav_player` both seem to play fine.

I have no other ESP32 board, so not at all exhaustive test.

dpgeorge, 2023-06-25:
We have now switched to IDF v5.0.2.  The above two fixes (defining DR_REG_I2S_BASE and updating the conditional to include C3) are still required.  I tested them with a GENERIC_C3 build on a ESP32-C3-DevKitC and I2S output works.

ma261065, 2023-07-22:
> Caveats: I tried the i2s examples `easy_wav_player` and `play_tone` by copying those files to the MCU- including wav files. I also tried with both binaries with the same results. I am using a [...]

I wonder if the noise issue is the same one that I discovered in https://github.com/micropython/micropython/issues/11245

Can you try the fix there and see if it solves the noise issue?

ricksorensen, 2023-07-22:
Thanks so much for this pointer- I haven't done exhaustive testing but setting the `i2s_config` parameters directly when TDM is available fixed the problem in the small test runs I made using the SEEED XIAO ESP32C3.  I was able to start and play stereo and mono wav files everytime using Mike Teachman's `easy_wav_player` and a generic PCM5102 board.

My current modifications using ESP IDV v5 are:  (all in `machine_i2s.c`)

- I include `driver/i2s_std.h` instead of `driver/i2s.h`  and I omit explicitly including `soc/i2sreg.h`. 
- I removed (commented out) the `REG_SET_BIT` macro definitions.  And since they are not included I no longer needed to `#define  DR_REG_I2S_BASE  0x6002D000`
- (Note that if you leave the macro definitions in `CONFIG_IDF_TARGET_ESP32C3` should be added to conditional, and the `DR_REG_I2S_BASE` definition added, and sound behavior is good)
- I added your `i2s_config`  updates immediately after setting `bits_per_chan = 0` in `machine_i2s_init_helper()` .

I still get a ESP32 message (not micropython)  when restarting I2S 
```
>>> import easy_wav_player
I2S(id=0,
sck=6,
ws=7,
sd=3,
mode=5,
bits=16, format=1,
rate=16000, ibuf=40000)
**E (87244) i2s(legacy): i2s_driver_uninstall(1494): I2S port 0 has not installed**
I2S(id=0,
sck=6,
ws=7,
sd=3,
mode=5,
bits=16, format=0,
rate=16000, ibuf=40000)
>>> 
**E (114214) i2s(legacy): i2s_driver_uninstall(1494): I2S port 0 has not installed**
MPY: soft reboot
MicroPython v1.20.0-311-gea1a5e43d-dirty on 2023-07-22; ESP32C3 XIAO with ESP32C3
Type "help()" for more information.
>>> 
``` 
but this doesn't seem to impact performance.  The first message is when switching I2S from stereo to monitor and changing frequency, the second is after a soft reset (I assume clearing the existing I2S instance)

Thanks again-

MrTheRich, 2024-05-16:
It's been 2 years since this was mentioned, I2S is still not included in the main firmware build. What problem is keeping this issue from being resolved? The C3 is getting more and more popular, it's weird that this module that has been ported already for the other esp32 ports is still missing here.

ricksorensen, 2024-05-17:
Try the latest development release for the C3, which includes I2S.  The next official release (1.23?) should have it.  See pull request #13727 from early March this year which is now merged into main code base.

MrTheRich, 2024-05-17:
@ricksorensen thanks for the response. I had hope for a second, but I tried the latest preview build of 1.23.0 preview 379 (16 may 2024) from the micropython website.
But it still does not seem to be in there:
[image]

MrTheRich, 2024-05-17:
Could this be it? Maybe this line should be changed, otherwise it is still not going to be included in the next build.

https://github.com/micropython/micropython/blob/cfd5a8ea3a80930a79dbae6a6aff54587393d0b8/ports/esp32/boards/ESP32_GENERIC_C3/mpconfigboard.h#L7

ricksorensen, 2024-05-17:
Yes indeed.  I had forgotten that I have this defined by default in my case (using a SEEED XIAO ESP32C3) 

`mpconfigboard.h`

``` c++
// This configuration is for a generic ESP32C3 board with 4MiB (or more) of flash.
//   modified for my ESP32C3

#define MICROPY_HW_BOARD_NAME               "ESP32C3 XIAO"
#define MICROPY_HW_MCU_NAME                 "ESP32C3"

#define MICROPY_HW_ENABLE_SDCARD            (0)
#define MICROPY_PY_MACHINE_DAC              (0)
#define MICROPY_PY_MACHINE_I2S              (1)
```

MrTheRich, 2024-05-18:
@ricksorensen I've been struggling for 2 days now to get this to build with the latest dev branch  and run, with the machine_I2S flag enabled. I finally got a firmware out of the build processes, but after flashing, it just goes in a boot loop, mentioning a load access fault.

With the new I2S version what all was necessary to get it to run?  Still all you mentioned a year ago?

I hoped it was just a simple case of changing the code and rebuilding, possibly even submitting a pr. It's very possible I'm overlooking something completely unrelated. Because even just getting the source to build was a multi hour head inducing struggle (not helped by not being very familiar with Linux)

If it's not to much to ask, could I try out your firmware build? Setting up audio with the esp32-c3 was supposed to be a small part of this project I'm working on, but it's already become a 3 day frustrating rabbit hole. And I've gotten no further on the rest of the project. If you have the time, my email is on my profile.

ricksorensen, 2024-05-18:
I'll try to get a version of my firmware to you, possibly tonight.  I'll also try to document my build process better.

I build with IDF v5.04 - if I use v5.2 my builds sometimes work but I have not successfully gotten them to start.

ricksorensen, 2024-05-18:
Here is a path to some of my examples (largely from Mike Teachman's notes and repositories):
[samples/esp32c3](https://github.com/ricksorensen/micropython-fun/tree/main/samples/esp32c3)

Here is may XIAO ESP32C3 binary: built with IDF 5.04
[SEEED_XIAO_ESP32C3_I2S.bin.zip](https://github.com/micropython/micropython/files/15366207/SEEED_XIAO_ESP32C3_I2S.bin.zip)

jonnor, 2024-07-07:
If I2S works with the MICROPY_PY_MACHINE_I2S set to 1 for the ESP32-C3, please provide a merge request - and the chances for it being included in the next versions goes up dramatically :)

ricksorensen, 2024-07-07:
I think this is covered in current release via the ESP IDF 5.2 at least.  See:
`ports\esp32\mpconfigport.h': `#define MICROPY_PY_MACHINE_I2S              (SOC_I2S_SUPPORTED)`

and from the esp32 tools directory:

```
 find . -name '*.h' | xargs grep SOC_I2S_SUPPORTED
./components/soc/esp32c6/include/soc/soc_caps.h:#define SOC_I2S_SUPPORTED               1
./components/soc/include/soc/i2s_periph.h:#if SOC_I2S_SUPPORTED
./components/soc/include/soc/i2s_periph.h:#if SOC_I2S_SUPPORTED
./components/soc/include/soc/i2s_periph.h:#endif // SOC_I2S_SUPPORTED
./components/soc/esp32h2/include/soc/soc_caps.h:#define SOC_I2S_SUPPORTED               1
./components/soc/esp32s3/include/soc/soc_caps.h:#define SOC_I2S_SUPPORTED               1
./components/soc/esp32p4/include/soc/soc_caps.h:#define SOC_I2S_SUPPORTED               1
./components/soc/esp32c3/include/soc/soc_caps.h:#define SOC_I2S_SUPPORTED               1
./components/soc/esp32/include/soc/soc_caps.h:#define SOC_I2S_SUPPORTED           1
./components/soc/esp32s2/include/soc/soc_caps.h:#define SOC_I2S_SUPPORTED               1
./components/hal/include/hal/i2s_types.h:#if SOC_I2S_SUPPORTED
./components/hal/include/hal/i2s_hal.h:#if SOC_I2S_SUPPORTED
./components/hal/include/hal/i2s_hal.h:#if SOC_I2S_SUPPORTED
./components/hal/include/hal/i2s_hal.h:#endif // SOC_I2S_SUPPORTED
```