micropython/micropython #8642: all python scripts used in the build should not depend on the current working directory

schneidersoft, 2022-05-07:
When python scripts are called from outside of their normal working directory they will not work as expected.

For example, micropython is used as a submodule in a larger project. The makeversionhdr.py script is called by a build system from outside of the micropython repo. the generated header will give the git hash of the parent repo, and not the micropython submodule...

instead these scripts could detect their own location and use the git info from the repo they are contained in.

dlech, 2022-05-07:
> the generated header will give the git hash of the parent repo, and not the micropython submodule

This is actually a "feature" that we use. All of our configs and extensions are outside of the MicroPython directory, so when we build MicroPython, the top level git directory commit has is the one that really says what the version is.

schneidersoft, 2022-05-07:
out of curiosity how are you building your application?

dlech, 2022-05-07:
The repo is here: https://github.com/pybricks/pybricks-micropython

The `bricks/` folder contains the equivalent of MicroPython ports/variants. We have our own top level makefiles that include some things from MicroPython. For example, https://github.com/pybricks/pybricks-micropython/blob/master/bricks/stm32/stm32.mk is based on the MicroPython minimal port.

schneidersoft, 2022-05-07:
Wow. Great stuff.

I'll be using this 'feature' as well then ;)

closed 2022-05-07 by schneidersoft