micropython/micropython #8643: Reduce flash storage of properties

jepler, 2022-05-08:
Over in CircuitPython we took a patch from me to reduce the flash storage size of properties that only have a getter, or only a getter and a setter. (Properties with just 1 or 2 slots are placed together in flash, so that address comparisons can reveal the number of slots)

As written, it's intended to be optional but I didn't measure whether it's truly zero-impact when disabled. It also affects the ABI (or is mp_obj_property_get not part of the ABI?)

If there is interest, I will prepare a version for Micropython. However, I wanted to ask first, as I know you don't prioritize the use of properties within core code, and it doesn't help with runtime properties. Feel free to close this up if that's the consensus on your end.

For more details, check out https://github.com/adafruit/circuitpython/pull/6337

jimmo, 2022-05-09:
Thanks @jepler -- Nice one! I considered something similar for the mp_obj_type_t slots thing, where you could just have a variable length set of pointer slots, and the "unary op" one would be the one where the corresponding function was in the "unary op region". But the additional cost of searching seemed prohibitive (although I didn't measure it). I think this technique could most likely be applied in other places though.

It occurred to be I couldn't even remember how to define a property in C, turns out we don't do this anywhere in MicroPython... so unfortunately it will be a net code size increase for us (although we can just leave the macro disabled).

For consistency, do you think it would be worth having a single MP_DEFINE_CONST_PROPERTY that took a variable number of arguments and that way you could use the same thing everywhere. Adapting the thing from #7542 would give you.

```
#define _MP_DEFINE_CONST_PROPERTY_0(n, g) mp_property_get_t n = {g}
#define _MP_DEFINE_CONST_PROPERTY_1(n, g, s) mp_property_getset_t n = {g,s}
#define _MP_DEFINE_CONST_PROPERTY_2(n, g, s, d) mp_property_getsetdel_t n = {g,s,d}

#define MP_DEFINE_CONST_PROPERTY_EXPAND(x) x

#define _MP_DEFINE_CONST_PROPERTY_EXPAND_IMPL(_1, _2, _3, _4, N, ...)  _MP_DEFINE_CONST_PROPERTY_##N
#define MP_DEFINE_CONST_PROPERTY(...) MP_DEFINE_CONST_PROPERTY_EXPAND(_MP_DEFINE_CONST_PROPERTY_EXPAND_IMPL(__VA_ARGS__, 2, 1, 0, _INV, _INV)(__VA_ARGS__))

int a, b, c;

MP_DEFINE_CONST_PROPERTY(z, a);
MP_DEFINE_CONST_PROPERTY(z, a, b);
MP_DEFINE_CONST_PROPERTY(z, a, b, c);
```

jepler, 2022-05-09:
The macro idea is interesting! Thanks for mentioning it. I think CP doesn't need it for now since the existing users have been converted but I can see where it would be advantageous.