micropython/micropython #8646: Pyboard UART.sendbreak

skylin008, 2022-05-10:
I used the UART1 on pyboard,the baudrate is 19200bps, when used the UART.sendbreak() function, the 13 bit delay is 677uS, but actual is 520uS in oscillograph.How to solve this issue.Thanks!

robert-hh, 2022-05-10:
Looks like it is sending a Null byte instead of a break. I failed at the same point trying to implement sendbreak() for SAMD. If timing is not overly critical, you can as a provisional workaround reduce the baudrate, send a NULL-Byte, and raise the baud rate again, like:
```
def my_sendbreak(baud):
    uart.init(baudrate=baud * 3 // 4)
    uart.write(b'\x00')
    uart.init(baudrate=baud)
```

skylin008, 2022-05-10:
Thank you @robert-hh ,I used the LIN BUS driver,I will be try.

dpgeorge, 2022-05-19:
The UART hardware is sending out 10 "0" bits, which is 520us at 19200 baud.  So it is correct.  The 10 bits are 8 data bits, start and stop.

robert-hh, 2022-05-19:
The sendbreak state must be different from sending a '0' byte. Therefore the length must be at least 13 bits which is 677µs at 19200 baud and 8 bit data.
b.t.w (and to check). At 9 bit data, with parity, start and 2 stop bits, one gets also 13 bits. In that case the break state must be even longer.

dpgeorge, 2022-05-19:
> The sendbreak state must be different from sending a '0' byte

I'm not sure what the official definition of a break is, but the `sendbreak()` method is just providing an interface to the underlying hardware.  Perhaps the hardware doesn't work as it should...

robert-hh, 2022-05-19:
I did not find a standard defining break, except some historical background relating to teletype machines, where "break" really meant "open current loop". (I had an ASCII teletype as a terminal to my first handwired Z80 board. Later also a classical 75 baud one. Got me much trouble with the other people in the apartment, because that one was really noisy, especially with the shift.)

When adding the UART method for SAMD, I noticed, that even when programming the controller to create the break state, it has send just a NULL byte. But while writing this I noticed that I could try it another way.